m5stack/M5Core2 #7: touch doesn't use interrupt wire

ropg, 2020-09-08:
Is there a reason why the reading of the touch chip interrupt line is commented out here? Seems to burn cycles when nothing is pressed.

```c++
bool touch::ispressed()
{
    //bool touch;
    uint16_t pos_x,pos_y;
    //touch = ( digitalRead(39) == LOW ) ? true : false;
    if(  readTouchtoBuff(&pos_x,&pos_y) == 0x02 )
    {
        return true;
    }
    else
    {
        return false;
    }

    return false;
}
```

lovyan03, 2020-09-09:
The INT on the FT6336U gives a pulsed signal by default.
By manipulating the register of the FT6336U, switch between LOW and HIGH in conjunction with touch.
I solved that with pull request #3 .

ropg, 2020-09-14:
Yeah. I wrote a whole new version of the touch driver that supports two touches as well as a TouchButton library to provide Arduino style buttons. Hope to incorporate gestures soon. See PR https://github.com/m5stack/M5Core2/pull/9.

(I was thinking for a moment of making the touch driver an interrupt driven, but that can't be done as it has no way of knowing what else is going on on that I2C bus.)

closed 2020-09-14 by ropg