m5stack/M5Core2 #15: Error occurs during compilation in Touch.h

AdalbertSZ, 2020-10-27:
An error occurs during compilation

_In file included from C:\Users\wszaj\Dysk Google\Arduino\FT6336U\Touch\Touch.ino:1:0:
C:\Users\wszaj\Dysk Google\Arduino\libraries\FT6336U/FT6336U.h:502:2: error: 'Finger' does not name a type

Finger _finger[2];

  ^_

tobozo, 2020-10-27:
`C:\Users\wszaj\Dysk Google\Arduino\FT6336U\Touch\` does not seem to be part of M5Core2 folder structure.

You can't expect the example sketch from an external driver to work natively with M5Core2 code.

You should try the example from the M5Core2 group instead:

[image]

Zontex, 2020-11-12:
Hi, as @tobozo mentioned this is not part of our library and therefore won't work, you can try the examples in the library to work your way around them.

closed 2020-11-12 by Zontex