m5stack/M5Core2 #17: DeepSleep function shuts system completely down and never puts ESP32 into deep sleep

felmue, 2020-11-10:
Hi guys

I don't think the DeepSleep function works as expected. When I call `M5.Axp.DeepSleep(1000 * 1000 * 2);` I'd expect the system to restart from deep sleep after about 2 seconds, but instead the system shuts down completely and never wakes up again from deep sleep.

The reason I think is that `SetSleep()`, called from within `DeepSleep()`, turns off power to the ESP32 IC and therefore line `(time_in_us == 0) ? esp_deep_sleep_start() : esp_deep_sleep(time_in_us);` is never executed and the system never goes into deep sleep but is shutdown completely.

Note: If I comment out `SetSleep()` then deep sleep works as expected.

Thanks
Felix

Zontex, 2020-11-12:
Hello, could you please provide the complete code you are trying to execute?

felmue, 2020-11-12:
Hello @Zontex 

sure, here you are. The code below never actually goes into deep sleep and therefore doesn't wake up after 2 seconds.
```
#include <M5Core2.h>

void setup() {
  M5.begin();

  Serial.println("Deep sleep for 2 seconds.");
  delay(100); // allow print to finish

  M5.Axp.DeepSleep(1000 * 1000 * 2);
}

void loop() {
}
```
I don't think commenting out `SetSleep()` is the correct solution here. I think `SetSleep()` is intended to prepare for deep sleep / light sleep and turn off everything that is not essential to save energy during deep sleep, e.g. everything except power to the ESP32.

BTW: M5StickC uses the same AXP192 and `SetSleep()` there is coded a bit differently. Maybe have a look there?

Thanks
Felix

felmue, 2020-11-15:
Hi @Zontex 

Re SetSleep(): I am wondering about below line - I don't think it does what the comment implicates:

`Write1Byte(0x31, Read8bit(0x31) | ( 1 << 3)); // short press to wake up`

Register 0x31 deals with power off voltages and (1 << 3) which equals to bit 3 is reserved in register 0x31 so in my opinion this line does nothing at all. Or am I missing something?

Thanks
Felix

Zontex, 2020-11-24:
> Hi @Zontex
> 
> Re SetSleep(): I am wondering about below lines:
> 
> ```
> uint8_t buf = Read8bit(0x31);
> buf = (1 << 3) | buf;
> Write1Byte(0x31, buf);
> ```
> 
> Register 0x31 deals with power off voltages and (1 << 3) which equals to bit 3 is reserved in register 0x31 so in my opinion this lines do nothing at all. Or am I missing something?
> 
> Thanks
> Felix

Hi @felmue does it has anything to do with your pull request regarding the AXP192? I've sent it to our engineers to take a look hopefully they will confirm it ASAP and I will merge your pull request, really appreciate your contribution!

If the pull request you've submitted doesn't solve this issue, I'll personally take a look into it to see what's going on.

felmue, 2020-11-24:
Hello @Zontex 

you are welcome, but no, my recent pull request regarding AXP192 doesn't solve this (deep sleep) issue. It corrects the RTC backup battery charging current which is not related to deep sleep.

Thank you for taking the time to look into this (deep sleep) issue. If I knew a solution I'd propose a pull request, but I am not clear myself what should go into `SetSleep()` function to prepare for deep sleep.

Thanks
Felix

felmue, 2020-11-25:
Hello @Zontex 

ok, today I took a stab at fixing light / deep sleep for M5Core2. I also added an example to show how light / deep sleep works.

Please have a look and please let your engineers double-check I didn't miss / broke anything. [Pull request](https://github.com/m5stack/M5Core2/pull/19)

Thanks
Felix

Zontex, 2020-11-26:
Hi @felmue sorry for the late reply, had a day off yesterday.
I'm super thankful for your contribution to the community, makes the world a better place for all of us.
I'm going to test it today and if no issues found I'll approve the pull request.
again, thanks for your contribution!

Zontex, 2020-11-26:
Hi @felmue I merged it into our master repo, thanks for your contribution!

closed 2020-11-26 by Zontex

felmue, 2020-11-26:
Hello @Zontex 
no worries, I hope you had a great day off.
Cheers
Felix