m5stack/M5Core2 #20: function AXP192::ScreenBreath set wrong register

derallye, 2020-11-26:
I guess function "ScreenBreah" is intended to control the LCD brightness, via DCDC3. 
AXP192 DC-DC3 control register is 27h, not 28h, as per AXP datasheet 1.13. If so, function writes to a wrong register.
As well, it is a completelly dedicated register, so brightness values can go from 0 to 127, so they must be limited the maximum allowable voltage (3V?), that means value 92.

----
void AXP192::ScreenBreath(uint8_t brightness)
{
    if (brightness > 12)
    {
        brightness = 12;
    }
    uint8_t buf = Read8bit(0x28);
    Write1Byte(0x28, ((buf & 0x0f) | (brightness << 4)));
}

trunneml, 2022-01-18:
`AXP192::ScreenBreath` changes LDO2, which is main power for the LCD and corresponding to the comment in line 42 also the voltage for the sdcard.
The ILI9342C has a buildin backlight logic, so I think there is no need at all to change the voltage of the backlight.

Tinyu-Zhao, 2022-07-26:
Thanks for the reminder, I have fixed the problem.

closed 2022-07-26 by Tinyu-Zhao