m5stack/M5Core2 #31: ECG example not working in Core2

iPsych, 2021-01-15:
It seems ECG example is not working in Core2.
Adjusted Serial lines doesn't work like described.
Is there more modification?

felmue, 2021-01-17:
Hello @iPsych 

I don't have the ECG module, but I think the GPIOs should be changed from 13, 5 to 19, 33.
Eg. try to modify this line:
`Serial1.begin(115200,SERIAL_8N1,13,5);`
to
`Serial1.begin(115200,SERIAL_8N1,19,33);`

As a reference - [the differences of the M5Stack bus and M5Core2 bus](https://www.gwendesign.ch/kb/m5stack/m5core2/#m5core2-bus-and-gpios-compared-to-m5stack).

Good luck!
Felix

iPsych, 2021-01-22:
@felmue 
Thanks.
It was the problem! The correct pin allows to run the example in Core2.

closed 2021-01-22 by iPsych