m5stack/M5Core2 #32: LCD functions don't update buffer but instead update screen directly

KarateBrot, 2021-01-18:
Whenever I want to call `M5.Lcd.clear()` or `M5.Lcd.clearDisplay()`, it immediately refreshes the display, making the screen flicker.
Also, `M5.Lcd.display()` doesn't do anything at all.

Solution:
Make LCD drawing functions only update the buffer until ultimately `M5.Lcd.display()` is being called to refresh the LCD.

tobozo, 2021-01-18:
The screen buffer you're referring to is inside the TFT, the clear() method is just sending a full frame of zeros, and this will always flicker on 320x240 screens.

Use [Sprites](https://github.com/m5stack/M5Core2/blob/master/src/utility/Sprite.h) if you want to control your own framebuffer and avoid this flickering effect.

KarateBrot, 2021-01-18:
Why is the frame buffer not part of the Lcd graphics library? `clear()` and `clearDisplay()` are very poorly implemented. Sorry to say that.

tobozo, 2021-01-18:
> Why is the frame buffer not part of the Lcd graphics library? 

Because this is TFT, not an OLED.
Only OLED adressing requires the full image on every update, although this is very unefficient, it doesn't produce flickering as pixel color is 1bit depth and 8192 bytes (128x64) is not long to transfer even on an I2C bus.

On the other hand a TFT has no refresh rate and does not need to constantly receive pixel data to work.
A full screen of 16bits pixels represents 1228800 bits (= 153600 bytes) of data to send via SPI, this will take time even if the bus is running at 80MHz, and it will result in a flickering effect because this is not an OLED.

If you don't want a flicker, don't update the whole screen, instead use sprites and partial screen updates.

Sprite::fillSprite() along with Sprite.pushSprite() do exactly what you're looking for

KarateBrot, 2021-01-18:
Again, flickering wouldn't be a problem if the screen would just get updated once instead of updating it on every method call. But it's fine, I'll just extend the display library myself to be able to use the `display()` method.

tobozo, 2021-01-19:
M5Display implements [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) for both TFT and Sprites, I'm sure you'll find answers to why this choice was made on their repository.

KarateBrot, 2021-01-19:
I'm closing this issue now because it's not an issue of the M5Core2 lib but rather the TFT_eSPI lib.

closed 2021-01-19 by KarateBrot

tobozo, 2021-01-22:
so it seems the way to use the display() function would be as follows:

```C

void myRenderFunction()
{

  M5.Lcd.startWrite();

  while( some loop conditions are met )  {
    // do some repeated drawing stuff
    M5.Lcd.fillCircle( .... );
    M5.Lcd.fillRect( .... );
    M5.Lcd.drawString( .... );
    M5.Lcd.display(); 
  }

  M5.Lcd.endWrite();

}

```