m5stack/M5Core2 #33: Close this project and extend the original API instead.

KodeMunkie, 2021-01-22:
Improve the original M5Stack API ([https://github.com/m5stack/M5Stack]https://github.com/m5stack/M5Stack) or create a common API to support both the Core 1 and Core 2 instead.

**Why?**

1. You have two issue trackers, how will you easily share, address and fix common reported bugs/issues in both projects?
2.  You have two copies of identical source with redundant broken code in this one such as M5.Power. Getting M5Stack API fixed to support the Core 2 will concentrate your mind and make you aware of the issues like this instead of ignoring them.
3. It's good practice - any devs like me that need to support builds for both 2 devices now have to put #defines all over our code to swap libraries depending on which device is targeted. This is causing extra work for numerous developers downstream - its trade off between 1 developer (you) or many (us) - that's not developer friendly.
4. It's causing confusion for newbies - searching for M5Core brings back 2 APIs. 
5. Tests. You haven't got any which means you should write them. You can share them between both devices and save time rather than cut and paste and running two pipelines and fixing them in 2 different projects.
6. I'll provide code where I've had to work around incompatibilities in my projects to save you time - they're workarounds/best effort but it's something.

Apologies for being direct but a large chunk of my current development time is identifying and fixing compatibility issues to do with the above _and it's incredibly annoying_.

lienbacher, 2021-01-25:
I am in a similar boat in here. Nice hardware, but please also up your software game m5stack!

bluepuma77, 2021-02-05:
+1

It's silly things. You switch from Core to Core2 and suddenly lcd.setBrightness() doesn't work. Suddenly M5Img() can only support PNG and not JPG anymore.

sieren, 2021-02-06:
+1

This project uses some weird old version of TFT_eSPI which is pretty unusable (ie the upstream TFT_eSPI version supports way better JPEG decoding through TFT_eFEX). +1000 to everything @KodeMunkie said.

Pensive, 2021-02-16:
I completely agree; It's a little disingenuous to not support font smoothing in the library (for example) but have implemented it in the factory test code loaded on the device by default, for example. Wishing I had just bought an m5stack now...

KodeMunkie, 2021-02-22:
How about it then @ropg ? :)

tobozo, 2021-02-22:
hello guys, if all you need is a single core for many devices, check out the [ESP32-Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core), contributions and PR are welcome :-)

KodeMunkie, 2021-02-22:
@tobozo Please don't hijack this thread. The third party API you link to doesn't actually support the Core 2 and there don't appear to be any plans to.

KodeMunkie, 2021-02-22:
> hello guys, if all you need is a single core for many devices, check out the [ESP32-Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core), contributions and PR are welcome :-)

@tobozo Please don't hijack this thread. The third party API you link to doesn't actually support the Core 2 and there don't appear to be any plans to.

tobozo, 2021-02-22:
@KodeMunkie yes it does support M5Core2, thanks for your contribution I'll update the ReadMe now

PeterRill, 2021-07-25:
@tobozo Thanks, I will give it a try. 

I had massive problems with the M5Core2 library. A program with Bluetooth BLE client causes reboots then I just added M5.begin()... With a Fire and M5Stack library it works...

UT2UH, 2021-07-27:
Another attempt to combine M5Stack/M5Core/M5StickC+ cores - [M5StX](https://github.com/UT2UH/M5Stack/tree/M5StX). Includes @ropg M5Touch, M5Sound (unmerged) and updated TFT_eSPI with M5_Font_Creator support and unified LcdBrightness and PowerOFF control. Made mostly for my own M5ez/ezTime based project so no complete compatibility intended but can be added.

Tinyu-Zhao, 2024-07-04:
Currently we have M5Unified, which is compatible with many hosts. Thank you for your feedback.

closed 2024-07-04 by Tinyu-Zhao