m5stack/M5Core2 #34: Most examples dont work in Arduino IDE

kraus-wagner, 2021-02-01:
Most of the examples dont work without a change. The compiler produces always errors.
Example TouchView:

Arduino: 1.8.13 (Windows 10), Board: "M5Stack-Core2, Enabled, Default (2 x 6.5 MB app, 3.6 MB SPIFFS), 240MHz (WiFi/BT), 921600, None"
TouchView:16:39: error: 'DIR_RIGHT' was not declared in this scope
Gesture swipeRight("Swipe Right", 80, DIR_RIGHT, 30, true);
TouchView:17:39: error: 'DIR_DOWN' was not declared in this scope
Gesture swipeDown( "Swipe Down",  60, DIR_DOWN,  30, true);
TouchView:18:39: error: 'DIR_LEFT' was not declared in this scope
Gesture swipeLeft( "Swipe Left",  80, DIR_LEFT,  30, true);
TouchView:19:39: error: 'DIR_UP' was not declared in this scope
Gesture swipeUp(   "Swipe Up",    60, DIR_UP,    30, true);
TouchView:66:19: error: variable or field 'eventHandler' declared void
void eventHandler(Event& e) {
TouchView:66:19: error: 'Event' was not declared in this scope
TouchView:66:26: error: 'e' was not declared in this scope
void eventHandler(Event& e) {
TouchView:72:21: error: variable or field 'gestureHandler' declared void
void gestureHandler(Event& e) {
TouchView:72:21: error: 'Event' was not declared in this scope
TouchView:72:28: error: 'e' was not declared in this scope
void gestureHandler(Event& e) {
TouchView:66:19: error: variable or field 'eventHandler' declared void
void eventHandler(Event& e) {
TouchView:66:19: error: 'Event' was not declared in this scope
TouchView:66:26: error: 'e' was not declared in this scope
void eventHandler(Event& e) {
TouchView:72:21: error: variable or field 'gestureHandler' declared void
void gestureHandler(Event& e) {
TouchView:72:21: error: 'Event' was not declared in this scope
TouchView:72:28: error: 'e' was not declared in this scope
void gestureHandler(Event& e) {
C:\Users\xxx\Documents\Arduino\M5Core2-master\examples\Touch\TouchView\TouchView.ino: In function 'void setup_events()':
TouchView:80:6: error: 'class M5Core2' has no member named 'background'
M5.background.delHandlers();
TouchView:81:49: error: 'E_ALL' was not declared in this scope
uint16_t events              = (show_move)  ? E_ALL : (E_ALL - E_MOVE); // Show all events, or everything but E_MOVE? Controlled with A button.
TouchView:81:66: error: 'E_MOVE' was not declared in this scope
uint16_t events              = (show_move)  ? E_ALL : (E_ALL - E_MOVE); // Show all events, or everything but E_MOVE? Controlled with A button.
TouchView:82:6: error: 'class M5Core2' has no member named 'background'
M5.background.longPressTime  = (long_press) ? 500   : 0;                // Detect long presses (500mS) or not? Controlled with B button.
TouchView:83:6: error: 'class M5Core2' has no member named 'background'
M5.background.repeatDelay    = (key_repeat) ? 200   : 0;                // Repeat press events every 200mS or not? Controlled with the C button.
TouchView:84:6: error: 'class M5Core2' has no member named 'background'
M5.background.addHandler(eventHandler, events);
TouchView:84:28: error: 'eventHandler' was not declared in this scope
M5.background.addHandler(eventHandler, events);
C:\Users\xxx\Documents\Arduino\M5Core2-master\examples\Touch\TouchView\TouchView.ino: In function 'void setupGestures()':
TouchView:94:27: error: 'gestureHandler' was not declared in this scope
swipeRight.addHandler(gestureHandler, E_GESTURE);
TouchView:94:43: error: 'E_GESTURE' was not declared in this scope
swipeRight.addHandler(gestureHandler, E_GESTURE);
exit status 1

Tinyu-Zhao, 2022-07-26:
Can you update the latest development version with the Core2 dependencies and burn the most basic examples to try again?
Because the problem is too long ago, I will close this issue, if there are still problems you are welcome to continue to comment.

closed 2022-07-26 by Tinyu-Zhao

Clement-HU, 2024-02-19:
but higher versions of M5Core2 will cause issues with M5GFX!And M5Unified cannot support many functions(for example,`M5.Touch.ispressed()`)