m5stack/M5Core2 #35: What about Speaker Beep ?

bigboxthailand, 2021-02-06:


PatrikRindlisbacher, 2021-02-10:
I miss the beep too

RaSTuS26, 2021-04-04:
Come on guys, please employ/enable the Speaker functions, seriously need them.

47tuc, 2022-02-04:
Any progress on this yet? I really need a simple beep function.

Tinyu-Zhao, 2022-07-26:
Hi I have added the basic dingdong sound for you to call.
https://github.com/m5stack/M5Core2/commit/1d62f63e4518a5a3a5ac2c173a31c25f97502513

closed 2022-07-26 by Tinyu-Zhao

capedra, 2023-03-13:
@Tinyu-Zhao 
Couldn't you please teach us how to generate our own unsigned char C arrays of sounds/music to play there instead ??

canDry, 2023-10-09:
Found this issue when trying to find tone/beep for M5 Core2...  found this library https://github.com/Hagtronics/M5Stack-Core2/tree/main/Sound... and modified it so that the code can just be added to my c++ CORE2 app (no includes needed).  
Example:  soundsBeep(freq,duration,%volume) // soundsBeep(1500,150,60);

Add these includes:
```
#include <driver/i2s.h>
#include <math.h>

```
Add this code: _(NOTE!  Of course you can always make this into an include... this is just done for simplicity)_

// speakerBeep() Error Codes
#define CORE2_SOUNDS_OK                                 0
#define CORE2_SOUNDS_MEMORY_ALLOC_ERROR                 -1
#define CORE2_SOUNDS_I2S_WRITE_ERROR                    -2
#define CORE2_SOUNDS_I2S_DID_NOT_WRITE_ALL_BYTES        -3

int32_t soundsPlay(const unsigned char *snd, size_t length) 
{
  size_t bytes_written = 0;
  //esp_err_t err = i2s_write(SPEAKER_I2S_NUMBER, snd, length, &bytes_written, portMAX_DELAY);
  i2s_port_t port = (i2s_port_t) 0;
  esp_err_t err = i2s_write(port, snd, length, &bytes_written, portMAX_DELAY);
  
  // Check errors, or return OK
  if(err != ESP_OK)
    return CORE2_SOUNDS_I2S_WRITE_ERROR;

  if(length != bytes_written)
    return CORE2_SOUNDS_I2S_DID_NOT_WRITE_ALL_BYTES;

  return CORE2_SOUNDS_OK;
}

int32_t soundsBeep(uint16_t freqHz, uint16_t timeMs, uint8_t volume)
{
  size_t rate = 44100;  // Samples per second
  size_t bytes_written = 0;
  size_t length = 0;

  // Realistic Frequency Response is > 1000 Hz and < 5000 Hz
  if(freqHz < 1000)
    freqHz = 1000;

  if(freqHz > 5000)
    freqHz = 5000;

  // Time is > 100 mSec & <= 1000 mSec
  if(timeMs > 1000)
    timeMs = 1000;      
```

  if(timeMs < 100)
    timeMs = 100;

  // 100% is max volume
  if(volume > 100)
    volume = 100;
  
  // Allocate memory for PCM table
      // TODO: uint16_t *raw = (uint16_t *)ps_calloc(rawLength, sizeof(uint16_t));
  length = (rate * timeMs) / 1000;
  // jsm int16_t *raw = (uint16_t *)calloc(length, sizeof(uint16_t));
  uint16_t *raw = (uint16_t *)calloc(length, sizeof(uint16_t));
  
  if(raw == NULL)
    return CORE2_SOUNDS_MEMORY_ALLOC_ERROR;

  // Calculate maximum value from user supplied volume
  int16_t maxVal = (int16_t)( (float)INT16_MAX * (volume / 100.0) );
[... 375 lines cut ...]
                         -0.32556815445715753,
                         -0.3090169943749477,
                         -0.29237170472273627,
                         -0.2756373558169998,
                         -0.2588190451025207,
                         -0.24192189559966787,
                         -0.22495105434386534,
                         -0.20791169081775987,
                         -0.19080899537654467,
                         -0.17364817766693127,
                         -0.1564344650402311,
                         -0.13917310096006588,
                         -0.12186934340514811,
                         -0.10452846326765342,
                         -0.08715574274765832,
                         -0.06975647374412476,
                         -0.05233595624294437,
                         -0.034899496702500823,
                         -0.01745240643728445};

    int index = (int)deg % 360;
    return sinmap[index];
}

```