m5stack/M5Core2 #37: Nothing works after flashing non UI firmware

MisteryX, 2021-02-25:
Hi
I'm new in this exciting world of embedded development.
I have got a M5 Core2, and it works only when I flash "UI Flow Core2" or "Core2 factory test firmware" to it. If I'm flashing example apps through M5Burner or c++ sketches through Arduino 1.8.13 (cp21x and all nessesary libs installed, Board: M5Stack-Core2, UploadSpeed 115200, 240MHz CPU, PSRAM: Enabled, Partition Scheme: default (2x6.5Mb))
nothing works at all. I just get brick, and turning on\off, reboot buttons does nothing.
Could you help me to figure out, what is wrong?

deepcoder, 2021-04-15:
I perhaps got into a similar state. After messing around with some demos my m5Stack Core - Fire appeared to no accept any program uploads from Arduino IDE. All that would happen was the LED strips would both turn on to bright white. What I ended up doing was a flash erase:

```
esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART erase_flash
```
then I flashed the follow test app, which works correctly. First install the library NeoPixelBus to control the m5Core's sk6812 LEDs.

```
// NeoPixelFunLoop
// This example will move a trail of light around a series of pixels.  
// A ring formation of pixels looks best.  
// The trail will have a slowly fading tail.
// 
// This will demonstrate the use of the RotateRight method.
//

#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

const uint16_t PixelCount = 10; // make sure to set this to the number of pixels in your strip
const uint16_t PixelPin = 15;  // make sure to set this to the correct pin, ignored for Esp8266
const uint16_t AnimCount = 1; // we only need one
const uint16_t TailLength = 3; // length of the tail, must be shorter than PixelCount
const float MaxLightness = 0.2f; // max lightness at the head of the tail (0.5f is full bright)

NeoGamma<NeoGammaTableMethod> colorGamma; // for any fade animations, best to correct gamma

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
// for esp8266 omit the pin
//NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount);

NeoPixelAnimator animations(AnimCount); // NeoPixel animation management object

void SetRandomSeed()
{
    uint32_t seed;

    // random works best with a seed that can use 31 bits
    // analogRead on a unconnected pin tends toward less than four bits
    seed = analogRead(0);
    delay(1);

    for (int shifts = 3; shifts < 31; shifts += 3)
    {
        seed ^= analogRead(0) << shifts;
        delay(1);
    }

    // Serial.println(seed);
    randomSeed(seed);
}

void LoopAnimUpdate(const AnimationParam& param)
{
    // wait for this animation to complete,
    // we are using it as a timer of sorts
    if (param.state == AnimationState_Completed)
    {
        // done, time to restart this position tracking animation/timer
        animations.RestartAnimation(param.index);

        // rotate the complete strip one pixel to the right on every update
        strip.RotateRight(1);
    }
}

void DrawTailPixels()
{
    // using Hsl as it makes it easy to pick from similiar saturated colors
    float hue = random(360) / 360.0f;
    for (uint16_t index = 0; index < strip.PixelCount() && index <= TailLength; index++)
    {
        float lightness = index * MaxLightness / TailLength;
        RgbColor color = HslColor(hue, 1.0f, lightness);

        strip.SetPixelColor(index, colorGamma.Correct(color));
    }
}

void setup()
{
    strip.Begin();
    strip.Show();

    SetRandomSeed();

    // Draw the tail that will be rotated through all the rest of the pixels
    DrawTailPixels();

    // we use the index 0 animation to time how often we rotate all the pixels
    animations.StartAnimation(0, 66, LoopAnimUpdate); 
}

void loop()
{
    // this is all that is needed to keep it running
    // and avoiding using delay() is always a good thing for
    // any timing related routines
    animations.UpdateAnimations();
    strip.Show();
}
```

MisteryX, 2021-05-10:
Thanx, it helped a lot.

closed 2021-05-10 by MisteryX