m5stack/M5Core2 #38: floppybird.ino  'TouchZone' does not name a type

robojax, 2021-04-03:
Hello,
I am trying to run the flopy bird gama under /examples/game/ 
at this line `TouchZone bottomHalf(0,120,320,160);`
it stops with    **'TouchZone' does not name a type**
some file is missing. please help.
Regards

felmue, 2021-04-04:
Hello @robojax 

yes, it's broken. The M5Core2 library has changed after the game has been added and nobody cared to fix it.

In order to make it compile try the following:

- comment out these not used lines:
```
//TouchZone topHalf(0,0,320,120);
//TouchZone bottomHalf(0,120,320,160);
//Gesture swipeDown(topHalf, bottomHalf, "Swipe Down");

//TouchButton lt = TouchButton(0, 0, 160, 120, "left-top");
//TouchButton lb = TouchButton(0, 120, 160, 120, "left-bottom");
//TouchButton rt = TouchButton(160, 0, 160, 120, "right-top");
//TouchButton rb = TouchButton(160, 120, 160, 120, "right-bottom");
```
- modify this line in setup
```
//  M5.Touch.addHandler(eventDisplay,TE_TOUCH);
  M5.Buttons.addHandler(eventDisplay,E_TOUCH);
```
- modify this line
```
//void eventDisplay(TouchEvent& e) {
void eventDisplay(Event& e) {
```

Thanks
Felix

P.S. I would provide a pull-request with this changes, but since pull-requests are mostly ignored by M5Stack lately, I won't bother doing that.

robojax, 2021-04-04:
Hi Felix,
Thank you very much. I was able to download the code to Core2.
I appropriate it.  I receive the module from Banggood for review. I am going to post a video on my YouRube channel Robojax Soon.
https://youtube.com/robojaxTV
Regards

closed 2021-04-04 by robojax