m5stack/M5Core2 #39: Cannot control screen brightness

teastainGit, 2021-04-12:
I Cannot control screen brightness on Core2. Neither ***M5.Lcd.setBrightness(255); or ***M5.Axp.ScreenBreath(10); have any affect. The boot screen reports ***axp: lcd backlight voltage was set to 2.80v. I cannot change this by putting ***M5.lcd backlight (2900); in the sketch set-up...or editing the AXP192 .cpp.
Please help!
Terry from Canada

felmue, 2021-04-14:
Hello @teastainGit 

I use `M5.Axp.SetLcdVoltage()` on my M5Core2 successfully. Not sure, why it doesn't work for you.

When you lookup function `SetLcdVoltage` you'll see that it takes values between `2500` and `3300`.
```
void AXP192::SetLcdVoltage(uint16_t voltage)
{
    if (voltage >= 2500 && voltage <= 3300)
    {
        SetDCVoltage(2, voltage);
    }
}
```
You can do that in `setup()` after calling `M5.begin()` - no need to modify `AXP192.cpp`.
```
M5.begin();
M5.Axp.SetLcdVoltage(2700);
```
BTW: the comment `axp: lcd backlight voltage was set to 2.80v` is static. So even if you change the code to `M5.Axp.SetLcdVoltage(2900)` you'd get the same comment.
```
    SetLcdVoltage(2800);
    Serial.printf("axp: lcd backlight voltage was set to 2.80v\n");
```
Thanks
Felix

teastainGit, 2021-04-14:
Yes, thank you Felix, I’ll try that!
Rather disappointed that the boot message is static…defeats the purpose?
Again, cheers!

Cheers !
Terry Stainton

Yoga Teacher
(519) 729-1047  (Call or text)
***@***.*** ***@***.***>

teastainGit, 2021-04-14:
THAT DOES THE TRICK!!!!
Cheers !
Terry Stainton

Yoga Teacher
(519) 729-1047  (Call or text)
***@***.*** ***@***.***>

Tinyu-Zhao, 2021-09-02:
Using M5.Axp.SetLcdVoltage() is a correct choice

closed 2021-09-02 by Tinyu-Zhao