m5stack/M5Core2 #41: No power 5V on Port A (I2C)

MisteryX, 2021-05-10:
Hi everyone! I`m stu(a)ck in another problem with my Core2 ;). After C++ programming session with Wav file playing, I got disabled I2C Port A and my sensors did not work anymore in UI Flow 1.7.6/1.7.5. I measured voltage on 5V line and GND and found it is 0. Could anyone help me? Is it a programm controller issue and I can turn PortA on back or it is mulfunction of SY7088 or AXP192?

koverg70, 2024-01-13:
Hi @MisteryX ! I have the same problem. 5V completely "disappeard" on my m5stack core 2. No I2C even no 5V bus. 
3.3V and battery (4.1V) work fine.
What was the solution if any for your problem?

MisteryX, 2024-01-13:
I just flashed a new version of UI Flow and problem disappeared.

koverg70, 2024-01-13:
> I just flashed a new version of UI Flow and problem disappeared.

Yes, it worked for me too. Thanks!

closed 2024-07-04 by Tinyu-Zhao