m5stack/M5Core2 #42: vector class is undefined and an error occurs

nagasawa, 2021-06-10:
M5Display.h

append
#include <vector>

take4mats, 2021-07-03:
This fixes below error I saw while executing trivial hello world sketch.  This is too easy fix, so am I missing something???

```cpp
#include <M5Core2.h>

void setup() {
  M5.begin();
  M5.Lcd.setTextSize(3);
  M5.Lcd.print("Hello World");
}

void loop() {
}
```

error log:

```
~/Documents/Arduino/libraries/M5Core2/src/M5Display.h:117:12: error: 'vector' in namespace 'std' does not name a template type
       std::vector<DisplayState> _displayStateStack;
```

bhupiister, 2021-11-16:
I had the same problem, but in my case the issue was installation of wrong library. I was using M5 core2 but had installed library for M5 stack on Arduino IDE

Tinyu-Zhao, 2022-07-26:
Thanks, fixed

closed 2022-07-26 by Tinyu-Zhao