m5stack/M5Core2 #52: 1.0.8 just released - Compilation errors.

Sarah-C, 2021-08-13:
While I'm compiling FastLED for a sketch, I'm getting multiple vector errors inside M5Stack/ files.... like so:
1.0.7 is fine.

"C:\\Users\\sarah\\AppData\\Local\\Temp\\arduino_build_793257\\sketch\\FadeLED.ino.cpp.o"
In file included from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:116,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:8,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms.h:36,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.cpp:2:
C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:16:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.cpp:2:
F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:65,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.cpp:2:
F:\sarah\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/clockless_rmt_esp32.h:116,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms/esp/32/fastled_esp32.h:8,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/platforms.h:36,
                 from F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:52,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.ino:2:
C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:16:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
In file included from F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Core2.h:76,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.ino:1:
F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Display.h:117:12: error: 'vector' in namespace 'std' does not name a template type
       std::vector<DisplayState> _displayStateStack;
            ^~~~~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Display.h:117:7: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Display.h:11:1:
+#include <vector>
 
F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Display.h:117:7:
       std::vector<DisplayState> _displayStateStack;
       ^~~
In file included from F:\sarah\Documents\Arduino\libraries\M5Core2\src/M5Core2.h:78,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.ino:1:
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:806:15: error: 'vector' in namespace 'std' does not name a template type
   static std::vector<Button*> instances;
               ^~~~~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:806:10: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:735:1:
+#include <vector>
 #endif /* M5Stack_M5Core2 */
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:806:10:
   static std::vector<Button*> instances;
          ^~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:893:15: error: 'vector' in namespace 'std' does not name a template type
   static std::vector<Gesture*> instances;
               ^~~~~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:893:10: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
   static std::vector<Gesture*> instances;
          ^~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:950:8: error: 'vector' in namespace 'std' does not name a template type
   std::vector<EventHandler> _eventHandlers;
        ^~~~~~
F:\sarah\Documents\Arduino\libraries\M5Core2\src/utility/M5Button.h:950:3: note: 'std::vector' is defined in header '<vector>'; did you forget to '#include <vector>'?
   std::vector<EventHandler> _eventHandlers;
   ^~~
In file included from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.ino:2:
F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from F:\sarah\Documents\Arduino\libraries\FastLED\src/FastLED.h:65,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.h:4,
                 from F:\sarah\Documents\Arduino\sketches\M5StackCore2\LEDPrograms\FadeLED\FadeLED.ino:2:
F:\sarah\Documents\Arduino\libraries\FastLED\src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Using library M5Core2 at version 0.0.4 in folder: F:\sarah\Documents\Arduino\libraries\M5Core2 
Using library Wire at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\SPI 
Using library FS at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\FS 
Using library SD at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\SD 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\SPIFFS 
Using library FastLED at version 3.4.0 in folder: F:\sarah\Documents\Arduino\libraries\FastLED 
Using library HTTPClient at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\HTTPClient 
Using library WiFi at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\WiFi 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\sarah\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\1.0.8\libraries\WiFiClientSecure 
exit status 1
Error compiling for board M5Stack-Core2.

tobozo, 2021-08-13:
`#include <vector>` is missing in `M5Button.h` but it probably happens because you're using an old SDK version from the M5Stack Boards manager ( `~/Arduino15/packages/m5stack` ).

Does it still occur when using the `M5Stack-Core2` from the `ESP32 Arduino` Boards manager ?

tobozo, 2021-08-13:
see pr #53

Sarah-C, 2021-08-13:
Cool, thanks!

closed 2021-08-13 by Sarah-C