m5stack/M5Core2 #70: M5Core2 example script for USB_MAX3421E not working on M5 AWS Core2

Disruptive-Prototyping, 2021-11-04:
Looks like this example script here has just been copied from M5Stack repo leaving it in unsupported commands e.g. "M5.Power.begin" even after removing this command, the USB host board does not detect or activate data transmission to USB device. Could be possible bus hardware issues due to new GPIO pins for MISO and IO7/PA_SCL on Core 2?

felmue, 2021-11-05:
Hello @Disruptive-Prototyping 

yes, tree GPIOs used for SPI have changed between M5Stack and M5Core2.

M5Stack: GPIO5 : SS, GPIO17 : INT, GPIO18 : SCK, GPIO19 : MISO, GPIO23 : MOSI
M5Core2: **GPIO33** : SS, **GPIO14** : INT, GPIO18 : SCK, **GPIO38** : MISO, GPIO23 : MOSI

Not sure though how to change them in the example code. Maybe it can be changed somewhere in the library?

I've documented all GPIO changes [here](https://www.gwendesign.com/kb/m5stack/m5core2/#m5core2-bus-and-gpios-compared-to-m5stack).

Thanks
Felix

Disruptive-Prototyping, 2021-11-06:
Hi Felix, 

You're absolutely right, I did manage to make this work with the help or your diagram early this week. To make it work on Core2 you need to change the relvant pin mapping in the USB Host Shield Library 2.0 under the ESP32 defined pins. 

The documentation is a little lacking online for this. 
Patching it together, you need to change the mapping of the SPI pins in the following files of USB Host Shield Library 2.0: avrpin.h, Usbcore.h & usbhost.h

***USBcore.h***
#elif defined(ESP32)
typedef MAX3421e<P5, P17> MAX3421E; // ESP32 boards. 

becomes:

#elif defined(ESP32)
typedef MAX3421e<P33, P38> MAX3421E; // ESP32 boards. 

***avrpin.h***
// Pinout for ESP32 dev module

MAKE_PIN(P0, 0);
MAKE_PIN(P1, 1); // TX0
MAKE_PIN(P10, 10); // TX1
MAKE_PIN(P3, 3); // RX0
MAKE_PIN(P21, 21); // SDA
MAKE_PIN(P22, 22); // SCL
MAKE_PIN(P19, 19); // MISO
MAKE_PIN(P23, 23); // MOSI
MAKE_PIN(P18, 18); // SCK
MAKE_PIN(P5, 5); // SS
MAKE_PIN(P17, 17); // INT

becomes:

// Pinout for ESP32 dev module

MAKE_PIN(P0, 0);
MAKE_PIN(P1, 1); // TX0
MAKE_PIN(P10, 10); // TX1
MAKE_PIN(P3, 3); // RX0
MAKE_PIN(P21, 21); // SDA
MAKE_PIN(P22, 22); // SCL
MAKE_PIN(P38, 38); // MISO
MAKE_PIN(P23, 23); // MOSI
MAKE_PIN(P18, 18); // SCK
MAKE_PIN(P33, 33); // SS
MAKE_PIN(P14, 14); // INT

***usbhost.h***
#elif defined(ESP32)
typedef SPi< P18, P23, P19, P5 > spi;

becomes:

#elif defined(ESP32)
typedef SPi< P18, P23, P38, P33 > spi;

Thanks for your time, have a great day!