m5stack/M5Core2 #78: M5stack Core2 + SD card Same SPI Bus issue

bhupiister, 2022-01-03:
I am using M5Stack Core2 and it uses TFT_eSPI library. I am facing issues with god knows what.
I am using SD card and TFT_eSPI on same bus. SD card initializes properly and so does the display.
But sometimes i face issues sometimes while accessing SD card, and i get an error
[ 88619][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed

I read suggestion somewhere of  using flag `USE_HSPI_PORT` using `HSPI `but then it does not load the screen.
I have read somewhere that it is because of ILI9341 using half-duplex mode and causing error to SD card as they are on the same SPI bus.

SD card initializing script
`SD.begin(SS_PIN, SPI, 40000000);`

Please help if you can on this issue, ill really appreciate any kind of help.

Tinyu-Zhao, 2022-07-25:
Hi, sorry for the late reply, have you tried to execute the [TFcard sample program](https://github.com/m5stack/M5Core2/blob/master/examples/Basics/tfcard/tfcard.ino), there is no problem to turn on the LCD and read/write to TFcard in my case.

https://user-images.githubusercontent.com/44396641/180679887-8212f399-01c4-4a16-a1a1-d245ae26cdad.MOV

bhupiister, 2022-07-25:
The problem is when you are transferring data at high speed and refreshing screen as well. For example, showing accelerometer data on graph and storing data at the same time. Anyways the problem was solved using semaphores

closed 2022-07-25 by bhupiister