m5stack/M5Core2 #79: min and std::min

obemu, 2022-01-05:
## Problem
Can you please rename the macro **min** in **utility/In_eSPI.h** to **M5CORE2_min**, because otherwise i get compilation errors in PlatformIO. It seems that this happens because the usage of **std::min(a,b)** in the toolchain-xtensa32@2.50200.97 package does not use **std::min** from the cpp algorithm library but it tries to use the macro **min** from **utility/In_eSPI.h**.

## Complete error message:
***
In file included from .pio/libdeps/debug/M5Core2/src/M5Display.h:10:0,
                 from .pio/libdeps/debug/M5Core2/src/M5Core2.h:76,
                 from src/main.cpp:9:
/home/emanuel/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/streambuf.tcc: In member function 'virtual std::streamsize std::basic_streambuf<_CharT, _Traits>::xsgetn(std::basic_streambuf<_CharT, _Traits>::char_type*, std::streamsize)':
.pio/libdeps/debug/M5Core2/src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token
   #define min(a,b) (((a) < (b)) ? (a) : (b))
                    ^
/home/emanuel/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/streambuf.tcc: In member function 'virtual std::streamsize std::basic_streambuf<_CharT, _Traits>::xsputn(const char_type*, std::streamsize)':
.pio/libdeps/debug/M5Core2/src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token
   #define min(a,b) (((a) < (b)) ? (a) : (b))
                    ^
/home/emanuel/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/istream.tcc: In member function 'std::streamsize std::basic_istream<_CharT, _Traits>::readsome(std::basic_istream<_CharT, _Traits>::char_type*, std::streamsize)':
.pio/libdeps/debug/M5Core2/src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token
   #define min(a,b) (((a) < (b)) ? (a) : (b))
                    ^
/home/emanuel/.platformio/packages/toolchain-xtensa32@2.50200.97/xtensa-esp32-elf/include/c++/5.2.0/bits/sstream.tcc: In member function 'virtual std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::overflow(std::__cxx11::basic_stringbuf<_CharT, _Traits, _Alloc>::int_type)':
.pio/libdeps/debug/M5Core2/src/utility/In_eSPI.h:543:20: error: expected unqualified-id before '(' token
   #define min(a,b) (((a) < (b)) ? (a) : (b))
                    ^
Compiling .pio/build/debug/libdbb/M5Core2/utility/Sprite.cpp.o
*** [.pio/build/debug/src/main.cpp.o] Error 1
=========================================== [FAILED] Took 7.82 seconds ===========================================

Environment    Status    Duration
-------------  --------  ------------
debug          FAILED    00:00:07.817
====================================== 1 failed, 0 succeeded in 00:00:07.817 ======================================
The terminal process "platformio 'run', '--environment', 'debug'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

***

OekoSolveMG, 2022-04-20:
I have exactly the same issue with the ```#define min```, it causes the ```std::chrono``^ c++ base library to not work anymore as well.

jangleboom, 2022-05-21:
Same issue here, it causes problems with code from "tensorflow/lite/schema/schema_generated.h". Renaming like [suggested by obemu](https://github.com/m5stack/M5Core2/issues/79#issue-1094463535) helped.

Tinyu-Zhao, 2024-07-04:
It looks like this issue has been resolved.

closed 2024-07-04 by Tinyu-Zhao