m5stack/M5Core2 #90: Support TouchButton according to official documentation

alptbz, 2022-03-06:
https://docs.m5stack.com/en/api/core2/touch
describes a way to use TouchButton
but this is not included in this library. Isn't this the official library?
Where is TouchButton implemented?

Tinyu-Zhao, 2022-03-12:
https://github.com/m5stack/M5Core2/blob/master/src/M5Touch.cpp

closed 2022-03-12 by Tinyu-Zhao

JohSchillerEmnify, 2022-04-07:
This does not solve the issue. These are separate things you posted Touch and Button but the documentation clearly refers to TouchButton class. See  here: 
```cpp
  #include <M5Core2.h>

  TouchButton lt = TouchButton(0, 0, 160, 120, "left-top");
  TouchButton lb = TouchButton(0, 120, 160, 120, "left-bottom");
  TouchButton rt = TouchButton(160, 0, 160, 120, "right-top");
  TouchButton rb = TouchButton(160, 120, 160, 120, "right-bottom");

  void colorButtons(TouchEvent& e) {
    TouchButton& b = *e.button;
    M5.Lcd.fillRect(b.x, b.y, b.w, b.h, b.isPressed() ? WHITE : BLACK);
  }

  void dblTapped(TouchEvent& e) {
    Serial.println("--- TOP RIGHT BUTTON WAS DOUBLETAPPED ---");
  }

  void setup() {
    M5.begin();
    M5.Touch.addHandler(colorButtons, TE_BTNONLY + TE_TOUCH + TE_RELEASE);
    rt.addHandler(dblTapped, TE_DBLTAP);
  }

  void loop() {
    M5.update();
  }
```

It is not possible to used this example, since a TouchButton class does not exist in the library. It is my understanding that this is the official library refers to. So either the example in the documentation is wrong. Or this libary is missing the literal first functionality described in the Touch documentation found here https://docs.m5stack.com/en/api/core2/touch