m5stack/M5Core2 #92: TFT_BL appears set incorrectly

belm0, 2022-05-07:
TFT_BL is defined as -1

So when the Display init code tries to set up the TFT backlight, there is an error from the esp32 library.

```
E (1375) ledc: ledc_channel_config(369): gpio_num argument is invalid
```

https://github.com/m5stack/M5Core2/blob/9ef6a81070022d566c07a9c01718c02c1a29d7f3/src/utility/In_eSPI_Setup.h#L187

https://github.com/m5stack/M5Core2/blob/9ef6a81070022d566c07a9c01718c02c1a29d7f3/src/M5Display.cpp#L21-L24

OekoSolveMG, 2022-05-20:
This line even seems to cause a crash for me in the newest Arduino core version.

belm0, 2022-05-21:
Yes, the offending code is left over from the Core Basic library and should be removed.

The maintainers don't seem to care-- perhaps focus is now on the M5Unified library.

Sarah-C, 2022-08-03:
I asked the devs of tft_eSPI about the backlight issue, because awhile ago it was overdriving it on the M5Stack. I think it ended up as "will not fix", the idea being we set it up after tft_eSPI has been initialised.

It would appear decisions about that are still leaking into the M5Stack repositories.

Below is more details about it, and a very short and simple Core 2 brightness function that handles it as it should -safely, below 2.8v but lets you overdrive the LED to 3.3 if you want. The engineers of the Core 2 don't recommend it for more than an hour at a time though.

https://github.com/Sarah-C/M5Stack_Core2_ScreenBrightness

https://www.reddit.com/r/esp32/comments/pnwdfj/warning_for_people_using_tft_espi_as_a_stand/
https://github.com/Bodmer/TFT_eSPI/issues/226

Tinyu-Zhao, 2022-08-17:
The Core2 backlight is controlled directly via the DCDC3 channel of the AXP193, setting -1 in TFT_BL means that it is not directly connected to a pin.

belm0, 2022-08-17:
> The Core2 backlight is controlled directly via the DCDC3 channel of the AXP193, setting -1 in TFT_BL means that it is not directly connected to a pin.

Maybe the TFT_BL value is sent to multiple places.  It should not be passed to `ledcAttachPin()` when it is -1, because this causes an error message.

Tinyu-Zhao, 2022-08-17:
So far everything looks fine, thanks for your feedback

commit 2af1a422 (2022-08-17): fix #92
closed 2022-08-17 by Tinyu-Zhao