m5stack/M5Core2 #97: Release 0.1.4 no longer compiles

DavidSchinazi, 2022-08-02:
Release 0.1.4 no longer compiles for me in a project where 0.1.3 worked. The compilation error is pasted below. I suspect that this bug was introduced by 1d62f63e4518a5a3a5ac2c173a31c25f97502513 since that commit added `Speaker.h` and `Speaker.cpp`. I think the fix is to add `#include FreeRTOS.h` to the top of `Speaker.h` 

```
Compiling .pio/build/core2aws/lib64b/M5Core2/Speaker.cpp.o
In file included from /Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/driver/driver/i2s.h:30:0,
                 from .pio/libdeps/core2aws/M5Core2/src/Speaker.h:4,
                 from .pio/libdeps/core2aws/M5Core2/src/Speaker.cpp:1:
/Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/semphr.h:74:3: error: #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
  #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
   ^
In file included from /Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/semphr.h:77:0,
                 from /Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/driver/driver/i2s.h:30,
                 from .pio/libdeps/core2aws/M5Core2/src/Speaker.h:4,
                 from .pio/libdeps/core2aws/M5Core2/src/Speaker.cpp:1:
/Users/me/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos/freertos/queue.h:75:3: error: #error "include FreeRTOS.h" must appear in source files before "include queue.h"
  #error "include FreeRTOS.h" must appear in source files before "include queue.h"
   ^
```

DavidSchinazi, 2022-08-02:
In case it helps, here's my env:
```
[env:core2aws]
framework = arduino
upload_speed = 1500000
monitor_speed = 115200
debug_tool = olimex-arm-usb-ocd-h
platform = espressif32
board = esp32dev
board_build.partitions = huge_app.csv
lib_deps =
  FastLED
  m5stack/M5Core2
```

DavidSchinazi, 2022-08-02:
My apologies, the issue was on my end, I was running a very old toolchain. Everything works with an updated toolchain.

closed 2022-08-02 by DavidSchinazi

Tinyu-Zhao, 2022-08-02:
Thank you for your feedback, I believe we are now compatible with the old version. https://github.com/m5stack/M5Core2/pull/96/commits/b42ec3d5a85ae9a5c1c535dd9e8846487648237a#diff-5879c7a936912964d8dcb2402bda08aff68bfa722fc8b462b4480cb0cd19691d

DavidSchinazi, 2022-08-02:
Oh that's great, thank you!