m5stack/M5Core2 #98: M5 Core 2 bricked after using EasyLoader for thermal camera

firex726, 2022-08-17:
### Describe the bug

M5 Core 2 module was working normally out  of the box using the default firmware. 
I tried the example arduino code found here but it would not compile, after installing the needed boards and libraries. 

I then tried the "EasyLoader" for the thermal camera found on your product page "https://docs.m5stack.com/en/unit/thermal". It appeared to burn the image and reported success, but now the module is bricked. And I cannot connect to it from the ArduinoIDE nor the M5Burner utility. 

### To reproduce

ArduinoIDE 1.8.16 & EasyLoder & m5Burner

### Expected behavior

Arduino project should compile and load. 
EasyLoader should burn example to device. 

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-08-18:
EasyLoad is mostly only applicable to M5Core series products, and does not match Core2, it is possible that the AXP192 or the serial chip is burned out.

firex726, 2022-08-18:
OK so where does it say that on your site?
https://docs.m5stack.com/en/unit/thermal

I would think if this has the possibility of bricking a device then there should be a note or even some form of verification in the software to verify what is connected.

felmue, 2022-08-18:
Hello @firex726 

- you could try w/o battery. Remove the little board, labelled CORE 2, then remove the bottom and disconnect the battery. (Careful the battery is attached to the bottom w/ double-sided tape.)
- w/o the battery, you could try to force ESP32 into download mode. Disconnect USB, connect a wire from GPIO0 to GND, then connect M5Core2 via USB to your computer.

Good luck!

Thanks
Felix