m5stack/M5Core2 #99: No way to power off external peripherials

JasonPittenger, 2022-08-29:
### Describe the bug

I have a device that I want to operate in sleep mode most of the time.
I have a GPS module attached to EXT.PORTA that I don't need while in sleep mode.
https://shop.m5stack.com/collections/m5-sensor/products/mini-gps-bds-unit
I want to power down the GPS module when it is not needed.

According to the m5 schematic prints, the 5V on EXT.PORTA is powered by BUS_5V, which is supplied by U8.
U8 has an enable pin attached to AXP192 pin 40 (EXTEN).

In the APX192 datasheet, EXTEN can be controlled using register 10 bit 2.
However, there is no way to write to this register with the current library.
Can this be added as a new function?
AXP192::PowerOffPeripherials()
{
  Write1Byte(0x0A, Read8bit(0x0A) & 0XFB);
}
AXP192::PowerOnPeripherials()
{
  Write1Byte(0x0A, Read8bit(0x0A) | 0X02);
}

### To reproduce

In any environment, this is not possible.

### Expected behavior

When you call AXP192::PowerOffPeripherials();
I expect "BUS_5V" to turn off, which would turn off anything attached to "BUS_5V" to turn off.

When you call AXP192::PowerOnPeripherials();
I expect "BUS_5V" to turn on, which would turn on anything attached to "BUS_5V" to turn on.

Default power on of the m5 would remain the same, with "BUS_5V" being powered on.

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Additional context

The GPS unit consumes around 40mA of current.
When the unit is asleep, it's not reading the GPS data.
This is wasted power.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-08-30:
Thanks for the suggestion, I have added the function to control PortA. [6fb8bfe](https://github.com/m5stack/M5Core2/commit/6fb8bfe9fb379101f9ebcb493b01350b5c67df21)

By the way the address should be 0x10, because the data sheet is marked "REG 10H" which already represents hexadecimal, not decimal

closed 2022-08-30 by Tinyu-Zhao

JasonPittenger, 2022-08-30:
Excellent work!
Thank you!

JasonPittenger, 2022-09-24:
I'm having trouble turning the peripherals back on.
If I call M5.Axp.SetPeripherialsPower(false), the 5V bus turns off.
However, when I call M5.Axp.SetPeripherialsPower(true), the 5V bus does not turn back on.
If I call AXP192::begin(), it turns the 5V bus back on.

I discovered that after calling M5.Axp.SetPeripherialsPower(true), you have to call M5.Axp.SetBusPowerMode(0) for the power to come back on.

Tinyu-Zhao, 2022-09-25:
Thank you for your feedback, I have fixed the problem.[fcd7dfc](https://github.com/m5stack/M5Core2/commit/fcd7dfc39ed37351e4807b0a572b85a1333fe611)