m5stack/M5Core2 #101: V 0.1.4 Will not compile

ortlof, 2022-09-01:
### Describe the bug

Code Builds not with 0.1.4 Version. 
New I2S is not initialized with framework-arduinoespressif32@3.10006.210326

### To reproduce

Change build to 0.1.4 -> code Fails to build with Error:

`C:/Users/-/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/freertos/freertos/semphr.h:74:3: error: #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
  #error "include FreeRTOS.h" must appear in source files before "include semphr.h"`

### Expected behavior

Build runs with 0.1.2

### Screenshots

Error:

`Indexing .pio\build\m5stack-core2\libd34\libWiFiClientSecure.a
In file included from C:/Users/Vampix/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/driver/driver/i2s.h:30:0,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/Speaker.h:4,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/Speaker.cpp:1:
C:/Users/-/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/freertos/freertos/semphr.h:74:3: error: #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
  #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
   ^
In file included from C:/Users/Vampix/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/freertos/freertos/semphr.h:77:0,
                 from C:/Users/Vampix/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/driver/driver/i2s.h:30,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/Speaker.h:4,
                 from .pio/libdeps/m5stack-core2/M5Core2/src/Speaker.cpp:1:
C:/Users/-/.platformio/packages/framework-arduinoespressif32@3.10006.210326/tools/sdk/include/freertos/freertos/queue.h:75:3: error: #error "include FreeRTOS.h" must appear in source files before "include queue.h"
  #error "include FreeRTOS.h" must appear in source files before "include queue.h"
   ^`

### Environment

- OS:
- IDE &IDE Version: Platformio:
- Repository Version: Platformio.ini
`
[env:m5stack-core2]
platform = espressif32@3.5.0
board = m5stack-core2
framework = arduino
monitor_speed = 115200
upload_protocol = esptool
board_build.partitions = default_16MB.csv
lib_deps = 
	m5stack/M5Core2@0.1.2
	madhephaestus/ESP32Encoder @ ^0.9.2
	mathertel/OneButton@^2.0.3
    lvgl@^8.3.1
build_flags = 
     -D LV_CONF_INCLUDE_SIMPLE
     -I src
`

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-09-01:
I will check it.

Tinyu-Zhao, 2022-09-17:
I did have this problem when I changed the library dependency to 0.1.4 in PlatformIO, but if I pull the latest repository on Github, it doesn't report the error, I suggest you try pulling the Github repository as a local dependency and try again, if that's ok I will release 0.1.5 to fix the problem.

ortlof, 2022-09-18:
Hi Thanks with 0.1.5 it complies when local pulled.

Tinyu-Zhao, 2022-09-19:
Thank you for your feedback.

commit 3243098a (2022-09-19): Merge branch 'patch-1' and fix #101
closed 2022-09-19 by Tinyu-Zhao