m5stack/M5Core2 #103: Touch Handler in M5 docs but not in library code

ejagombar, 2022-09-21:
### Describe the bug

The M5 Core2 docs found here http://docs.m5stack.com/en/api/core2/touch say that a handler can be added to the touch screen to which will allow a function to be called whenever the screen is pressed. An example shows it being used as `M5.Touch.addHandler(colorButtons, TE_BTNONLY + TE_TOUCH + TE_RELEASE);` however this addHandler function is not part of the current Touch object. 

Whats going on here? Am I missing something or are the docs just out of date?

### To reproduce

Compare docs at http://docs.m5stack.com/en/api/core2/touch to the Touch object in M5Touch.cpp

### Expected behavior

The offical docs should match the code in the repo.

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

JasonPittenger, 2022-09-24:
I noticed that as well.  It seems the documentation is lagging behind the examples.
However, I was able to get it to work off the examples.

I'm not part of the development team, but I might be able to help you figure out some of how to use the library.

Tinyu-Zhao, 2022-10-01:
You can try this example

```
#include <M5Core2.h>

ButtonColors onCol  = {BLACK, WHITE, WHITE};
ButtonColors offCol = {RED, WHITE, WHITE};
Button myButton(10, 10, 200, 100, false, "I'm a button !", onCol, offCol);

void touched(Event& e) {
    Serial.println("Touched!");
}

void released(Event& e) {
    Serial.println("Released!");
}

void setup() {
    M5.begin();
    myButton.addHandler(touched, E_TOUCH);
    myButton.addHandler(released, E_RELEASE);
}

void loop() {
    M5.update();
}

```

closed 2023-11-09 by ejagombar