m5stack/M5Core2 #105: Core2 can get RTC interrupt signal?

LunarEclipse49, 2022-09-28:
Hello, all

I want to get millisec for logging some data with RTC.

I tried using millis() first.
But if using `ms = millis()%1000`, sometime happen problem like this.
```
...
10:10:5.750
10:10:5.950
10:10:5.150   <-
10:10:6.350
10:10:6.550
...
```

To solve this, I found the idea that using RTC interrupt from [this site](http://kawalabo.blogspot.com/2011/12/arduinortc8564.html) (sorry, this is in Japanese)
I remaking it for M5Core2, but I have no idea to get interrupt signal from BM8564.
Would you give me which pin can listen to BM8564?

felmue, 2022-09-28:
Hello @LunarEclipse49 

unfortunately the interrupt signal from the BM8564 is **not** connected to a GPIO of the ESP32. (It is only connected to the PEK of the AXP192 which allows the BM8564 to power on the M5Core2.)

Please have a look at the schematic [here](https://docs.m5stack.com/en/core/core2).

Thanks
Felix

LunarEclipse49, 2022-09-29:
I'm glad to get your reply, Felix.

Thank you for link to schematic.
Now I understanding it.

This is my first time to use such a device.
It was difficult to understand them before starting development...

closed 2022-09-29 by LunarEclipse49