m5stack/M5Core2 #106: Cannot install m5core2 in Arduino IDE 2.0.0

tiede, 2022-10-09:
### Describe the bug

The M5Core2 library cannot be installed in Arduino IDE. It fails silently due to a missing dependency.

The log says the following:

root ERROR Request listDependencies failed with error: 9 FAILED_PRECONDITION: No valid dependencies solution found

I have tried to install one of the dependencies manually - the M5Family library. It fails with the following log error:

root ERROR Request listDependencies failed with error: 9 FAILED_PRECONDITION: No valid dependencies solution found: dependency 'M5_BM8563' is not available

The M5_BM8563 library cannot be found in library manager.

### To reproduce

1. Using a clean Arduino IDE
2. Search for M5Core2 
3. Install it and it fails silently

### Expected behavior

That the library is installed.

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

framinlab, 2022-10-10:
Possibly related issue, but in Arduino 1.8.19. When going to install M5Core2: 

Library is already installed: ADXL345:0.2.0
Library is already installed: FastLED:3.5.0
 no protocol: 
java.lang.RuntimeException: java.net.MalformedURLException: no protocol: 
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:250)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.MalformedURLException: no protocol: 
	at java.net.URL.<init>(URL.java:611)
	at java.net.URL.<init>(URL.java:508)
	at java.net.URL.<init>(URL.java:457)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:60)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:56)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:148)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:118)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:240)
	... 1 more

Same issue when you go to update to the latest version of the library. 

On Arduino IDE 2.0 it fails silently.

Tinyu-Zhao, 2022-10-24:
fixed
[image]

closed 2022-10-24 by Tinyu-Zhao