m5stack/M5Core2 #109: A fatal error occurred: Failed to write to target RAM

konacurrents, 2022-10-19:
### Describe the bug

Uploading the Factory Test for the Core2 gets an error and I can't work around it. I've tried on my Mac with m1 as well as older Mac. 

Error:
```
esptool.py v3.3
Serial port /dev/cu.usbmodem54780264861
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 78:21:84:93:38:58
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
A fatal error occurred: Failed to write to target RAM (result was 01070000: Operation timed out)
```

I've tried the UCB-C to UCB-C as well as UCB-A to UCB-C cables with no difference.

### To reproduce

Arduino 1.8.19

Compile the FactoryTest and try to upload.

### Expected behavior

The upload to work.

### Environment

- OS: osx, maxOS Monterey 12.6, MacBook Air.   But also on an older MacBook.
- IDE &IDE Version: Arduino 1.8.19
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-10-24:
In my case it is normal, I can not reproduce this problem.
[image]

konacurrents, 2022-10-24:
I have the latest of all the libraries. I don't know what you mean by "M5Core2 Repository". 
I was going to try the latest arduino but on first use I haven't figured it out yet (like my serial monitor is goofed up).

So still no luck:

[image]

konacurrents, 2022-10-24:
This is the compilation showing the libraries:
```
/Users/scott/Library/Arduino15/packages/m5stack/tools/esptool_py/3.3.0/esptool --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 16MB -o /var/folders/j7/x8fmgg910wzd23nl477qt0gw0000gp/T/arduino_build_878103/FactoryTest.ino.bin /var/folders/j7/x8fmgg910wzd23nl477qt0gw0000gp/T/arduino_build_878103/FactoryTest.ino.elf
esptool.py v3.3
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
python3 /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/tools/gen_esp32part.py -q /var/folders/j7/x8fmgg910wzd23nl477qt0gw0000gp/T/arduino_build_878103/partitions.csv /var/folders/j7/x8fmgg910wzd23nl477qt0gw0000gp/T/arduino_build_878103/FactoryTest.ino.partitions.bin
Multiple libraries were found for "WiFi.h"
 Used: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
Using library ArduinoECCX08 at version 1.3.6 in folder: /Users/scott/Documents/Arduino/libraries/ArduinoECCX08 
Using library Wire at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/Wire 
Using library M5Core2 at version 0.1.5 in folder: /Users/scott/Documents/Arduino/libraries/M5Core2 
Using library SPI at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/SPI 
Using library FS at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/FS 
Using library SD at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/SD 
Using library SPIFFS at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/SPIFFS 
Using library FastLED at version 3.5.0 in folder: /Users/scott/Documents/Arduino/libraries/FastLED 
Using library WiFi at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/WiFi 
Using library HTTPClient at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/HTTPClient 
Using library WiFiClientSecure at version 2.0.0 in folder: /Users/scott/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.5/libraries/WiFiClientSecure 
/Users/scott/Library/Arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-size -A /var/folders/j7/x8fmgg910wzd23nl477qt0gw0000gp/T/arduino_build_878103/FactoryTest.ino.elf
Sketch uses 1418025 bytes (21%) of program storage space. Maximum is 6553600 bytes.
Global variables use 49716 bytes (1%) of dynamic memory, leaving 4472268 bytes for local variables. Maximum is 4521984 bytes.

```

Tinyu-Zhao, 2022-11-13:
@konacurrents I have tried to reproduce the problem again today, please try using the other port name "/dev/cu.wchusbserial...." instead of "/dev/cu.usbmodem ...."
[image]

Tinyu-Zhao, 2022-11-15:
> Thanks for following up. I just tried with the new Arduino 2.0.1 and at least I can connect to the device.
> 
> 
> 
> Now I get an error (right away) on compiling.
> 
> ```
> 
> 
> 
> Error resolving FQBN: getting build properties for board m5stack:esp32:m5stack-core2: invalid option 'EraseFlash'
> 
> 
> 
> Compilation error: Error resolving FQBN: getting build properties for board m5stack:esp32:m5stack-core2: invalid option 'EraseFlash'
> 
> ```
> 
> 
> 
> Here is a screen shot. I tried an example from M5Stack-Core2 as well and it gives same error right away.
> 
> Any ideas?
> 
> thanks.
> 
> [image]
> 
> 
> 
> ### PS. you know how to change the monitor colors?
> 
> In the older Arduino, the preferences.txt let me do this. But I cannot find the equivalent..
> 
> 
> 
> 

Currently there are some problems with Arduino IDE 2.0 series, please continue to use 1.8.19 for stability.

konacurrents, 2022-11-15:
Strange. I was finally able to connect to the core2 with the new Arduino 2.0. But that error shows up.  Do you have any idea on that error? 

I'll try with the older Arduino too.
thanks,

Tinyu-Zhao, 2022-11-16:
I think you can update the development installer to this version.
[image]

closed 2024-07-04 by Tinyu-Zhao