m5stack/M5Core2 #110: MP3 files play with "slow" speed

NDC-SC, 2022-10-20:
When MP3 files are played with M5Core2 0.1.3 and the Audio libraries, the sound quality is fine.
With the M5Core2 0.1.4 or 0.1.5 libraries, the MP3 files play with a "slow" speed.
Instructions are
  #include <Audio.h>
  Audio audio;
  temp = filenames(filenumber];   // the filenumber MP3 file on the micro-SD card
  audio.connecttoFS(SD, temp.c_str());   // MP3 file on micro-SD card
Is there a work around to resolve the playback speed problem ?

Tinyu-Zhao, 2022-10-23:
Can this happen with other equipment?

NDC-SC, 2022-10-23:
Only found the problem with the M5Stack Core2

Tinyu-Zhao, 2022-10-23:
In order to reproduce the problem as soon as possible, you can provide the source code of the problem, you can send it to my email:Tinyu.Zhao@gmail.com

NDC-SC, 2022-10-23:
sketch emailed

NDC-SC, 2022-10-29:
Tinyu Zhao: has the reason for the M5Stack Core2 playing MP3 files slowly with M5Core2 0.1.4 or 0.1.5 libraries been determined ?

peticsizmadia, 2022-11-08:
Hi @Tinyu-Zhao, @NDC-SC!

I have the same probelm on M5Stack Core2. I tried with 0.1.5 and 0.1.4 libraries, and with both I have the same error.
After I tried with 0.1.3, it work good, the music is normal.

I use that code:
```
#include <M5Core2.h>
#include <driver/i2s.h>
#include "AudioFileSourceSD.h"
#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2S.h"

AudioGeneratorMP3 *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

#define OUTPUT_GAIN 10

void setup()
{
  M5.begin();
  M5.Axp.SetSpkEnable(true);
  file = new AudioFileSourceSD("/test.mp3");
  id3 = new AudioFileSourceID3(file);
  out = new AudioOutputI2S(0, 0);
  out->SetPinout(12, 0, 2);
  out->SetOutputModeMono(true);
  out->SetGain((float)OUTPUT_GAIN/100.0);
  mp3 = new AudioGeneratorMP3();
  mp3->begin(id3, out);
}

void loop()
{
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    delay(1000);
  }
}
```

aliasgherman, 2022-12-21:
I can confirm that I faced the same issue with the M5Core2 libraries 0.14 and 0.15. 
It works fine with 0.13
The 0.14/0.15 donot have slow mp3 playback (downsampled) no matter if the CPU frequency is changed OR output sampling rate is modified.

Aardvarx55, 2023-03-10:
I am seeing the same issue with streaming also... I will try downgrading the library

NexusG17, 2023-03-25:
Same issue with the M5Core2 last version library
Works ok with 0.13

I use the same test code as provided by @peticsizmadia

Aardvarx55, 2023-03-25:
Thanks, I too have 'downgraded' to 0.13, and will stay with that until I
hear the issue is resiolved.

zsoltyfm, 2023-10-08:
Hey guys!
In the case of newer versions, the speaker must be disabled and the problem will be solved immediately.
`void setup() {
  //M5.begin();
  M5.begin(true, true, true, true, kMBusModeOutput, false);   //begin(bool LCDEnable = true, bool SDEnable = true, bool SerialEnable = true, bool I2CEnable = false, mbus_mode_t mode = kMBusModeOutput, bool SpeakerEnable = true);
`

NDC-SC, 2023-10-08:
Brilliant. Thanks for resolving the "slow playback speed" problem.

Aardvarx55, 2023-10-08:
Excellent, works a treat... many thanks. I can now use the latest version 0.1.6

closed 2024-07-03 by Tinyu-Zhao