m5stack/M5Core2 #112: THERMAL_MLX90640 Arduino example will not compile

firex726, 2022-11-01:
The example Arduino file for the M5Core2 unit will not compile.

[Source file](https://github.com/m5stack/M5Core2/tree/master/examples/Unit/THERMAL_MLX90640)

Error from Arduino IDE:

`
In file included from C:\Users\firex\Downloads\M5Core2-master (1)\M5Core2-master\examples\Unit\THERMAL_MLX90640\MLX90640_I2C_Driver.cpp:18:0:
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire\src/Wire.h: In function 'int MLX90640_I2CRead(uint8_t, unsigned int, unsigned int, uint16_t*)':
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire\src/Wire.h:103:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)
     uint8_t requestFrom(int address, int size);
             ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\Wire\src/Wire.h:101:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)
     uint8_t requestFrom(uint8_t address, uint8_t size);
             ^
In file included from C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:30:0,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.h:4,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp:1:
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/semphr.h:74:3: error: #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
  #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
   ^
In file included from C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/semphr.h:77:0,
                 from C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:30,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.h:4,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp:1:
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:75:3: error: #error "include FreeRTOS.h" must appear in source files before "include queue.h"
  #error "include FreeRTOS.h" must appear in source files before "include queue.h"
   ^
In file included from C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/semphr.h:77:0,
                 from C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:30,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.h:4,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp:1:
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:613:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueGenericSend( QueueHandle_t xQueue, const void * const pvItemToQueue, TickType_t xTicksToWait, const BaseType_t xCopyPosition ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:725:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueuePeekFromISR( QueueHandle_t xQueue, void * const pvBuffer ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:906:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueGenericReceive( QueueHandle_t xQueue, void * const pvBuffer, TickType_t xTicksToWait, const BaseType_t xJustPeek ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:917:1: error: 'UBaseType_t' does not name a type
 UBaseType_t uxQueueMessagesWaiting( const QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:930:1: error: 'UBaseType_t' does not name a type
 UBaseType_t uxQueueSpacesAvailable( const QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:940:43: error: expected initializer before 'PRIVILEGED_FUNCTION'
 void vQueueDelete( QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
                                           ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1267:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueGenericSendFromISR( QueueHandle_t xQueue, const void * const pvItemToQueue, BaseType_t * const pxHigherPriorityTaskWoken, const BaseType_t xCopyPosition ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1268:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueGiveFromISR( QueueHandle_t xQueue, BaseType_t * const pxHigherPriorityTaskWoken ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1347:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueReceiveFromISR( QueueHandle_t xQueue, void * const pvBuffer, BaseType_t * const pxHigherPriorityTaskWoken ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1354:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueIsQueueEmptyFromISR( const QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1355:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueIsQueueFullFromISR( const QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1356:1: error: 'UBaseType_t' does not name a type
 UBaseType_t uxQueueMessagesWaitingFromISR( const QueueHandle_t xQueue ) PRIVILEGED_FUNCTION;
 ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/queue.h:1374:1: error: 'BaseType_t' does not name a type
 BaseType_t xQueueAltGenericSend( QueueHandle_t xQueue, const void * const pvItemToQueue, TickType_t xTicksToWait, BaseType_t xCopyPosition );
[... 110 lines cut ...]
                                                                 ^
In file included from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.h:4:0,
                 from C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp:1:
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:390:85: error: 'TickType_t' has not been declared
 esp_err_t i2s_read(i2s_port_t i2s_num, void *dest, size_t size, size_t *bytes_read, TickType_t ticks_to_wait);
                                                                                     ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:408:61: error: 'TickType_t' has not been declared
 int i2s_push_sample(i2s_port_t i2s_num, const void *sample, TickType_t ticks_to_wait) __attribute__ ((deprecated));
                                                             ^
C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/driver/driver/i2s.h:426:54: error: 'TickType_t' has not been declared
 int i2s_pop_sample(i2s_port_t i2s_num, void *sample, TickType_t ticks_to_wait) __attribute__ ((deprecated));
                                                      ^
C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp: In member function 'bool Speaker::InitI2SSpeakOrMic(int)':
C:\Users\firex\OneDrive\Documents\Arduino\libraries\M5Core2\src\Speaker.cpp:17:11: error: 'I2S_COMM_FORMAT_STAND_I2S' was not declared in this scope
           I2S_COMM_FORMAT_STAND_I2S,  // Set the format of the communication.
           ^
Multiple libraries were found for "WiFiClient.h"
 Used: C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\firex\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board M5Stack-Core2.
`

Tinyu-Zhao, 2022-11-03:
Please update the M5Stack version to 2.0.4, it will work well in this version.

Low versions may cause incompatibility problems.

[image]

closed 2022-11-03 by Tinyu-Zhao

firex726, 2022-11-03:
Exactly what am I supposed to be updating? 

I have already downloaded the latest versions of M5Burner, UIflow and arduino libraries. And why would the hardware being out of date cause your software to not compile?

Tinyu-Zhao, 2022-11-03:
> Exactly what am I supposed to be updating?
> 
> I have already downloaded the latest versions of M5Burner, UIflow and arduino libraries. And why would the hardware being out of date cause your software to not compile?

[image]

[image]

firex726, 2022-11-03:
`Platform m5stack:esp32@2.0.5+1.0 already installed
`

Results in:

`
exit status 1

Compilation error: exit status 1`

firex726, 2022-11-03:
`Using board 'm5stack-core2' from platform in folder: C:\Users\firex\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.5+1.0
Using core 'esp32' from platform in folder: C:\Users\firex\AppData\Local\Arduino15\packages\m5stack\hardware\esp32\2.0.5+1.0
cmd /c if exist "C:\\Users\\firex\\Downloads\\M5Core2-master (1)\\M5Core2-master\\examples\\Unit\\THERMAL_MLX90640\\partitions.csv" COPY /y "C:\\Users\\firex\\Downloads\\M5Core2-master (1)\\M5Core2-master\\examples\\Unit\\THERMAL_MLX90640\\partitions.csv" "C:\\Users\\firex\\AppData\\Local\\Temp\\arduino-sketch-1965224AC4B08F3D9DE353C7264828B5\\partitions.csv"
cmd /c if not exist "C:\\Users\\firex\\AppData\\Local\\Temp\\arduino-sketch-1965224AC4B08F3D9DE353C7264828B5\\partitions.csv" if exist "C:\\Users\\firex\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\variants\\m5stack_core2\\partitions.csv" COPY "C:\\Users\\firex\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\variants\\m5stack_core2\\partitions.csv" "C:\\Users\\firex\\AppData\\Local\\Temp\\arduino-sketch-1965224AC4B08F3D9DE353C7264828B5\\partitions.csv"

cmd /c if not exist "C:\\Users\\firex\\AppData\\Local\\Temp\\arduino-sketch-1965224AC4B08F3D9DE353C7264828B5\\partitions.csv" COPY "C:\\Users\\firex\\AppData\\Local\\Arduino15\\packages\\m5stack\\hardware\\esp32\\2.0.5+1.0\\tools\\partitions\\default_16MB.csv" "C:\\Users\\firex\\AppData\\Local\\Temp\\arduino-sketch-1965224AC4B08F3D9DE353C7264828B5\\partitions.csv"
The system cannot find the file specified.
exit status 1

Compilation error: exit status 1`

Tinyu-Zhao, 2022-11-06:
Now, you can try the latest version 2.0.5_1.0