m5stack/M5Core2 #114: Where's the original firmware for M5Stack Core2?

tommy4111, 2022-11-22:
### Describe the bug

Can't find the original firmware with all the meters at.  I want to make some custom changes to the GUI.

### To reproduce

firmware

### Expected behavior

firmware

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-11-26:
Do you mean this firmware?
[image]

tommy4111, 2022-12-03:
This one.  The one that shipped with the device.

[image]

Tinyu-Zhao, 2022-12-03:
The factory test program is now compatible with Core2.
https://github.com/m5stack/M5Core2/tree/master/examples/FactoryTest

closed 2023-01-15 by Tinyu-Zhao