m5stack/M5Core2 #124: Custom i2s_write src

capedra, 2023-03-13:
@Tinyu-Zhao 
How's it possible to generate a custom array that is currently found here (https://github.com/m5stack/M5Core2/blob/master/examples/FactoryTest/FactoryTest/preview_R.c) to play custom sounds/music from my own instead of just ```M5.Spk.DingDong();``` or ```  i2s_write(SPAKER_I2S_NUMBER, previewR, 120264, &bytes_written, portMAX_DELAY);```

Thank you very much !!!

Tinyu-Zhao, 2023-03-13:
You can try using these tools.

https://github.com/m5stack/M5Stack/blob/master/tools/bin2code.py

https://github.com/imliubo/A-Eye-Image2array-tool

Tinyu-Zhao, 2023-04-02:
> Hello,
> 
> Take a look at this: [Sound on a M5Core2](https://github.com/Hagtronics/M5Stack-Core2/tree/main/Sound) Is it what you mean?
> 
> Regards Tom

You are giving examples related to playing sound

closed 2024-07-03 by Tinyu-Zhao