m5stack/M5Core2 #126: CORE2 stack overflow

fbritop, 2023-03-26:
I had no trouble with previous devices from CORE2 (2 test units), I did order 20 units to proceed, but none of them work.
I have donde even a very simple Arduino Program with no luck

```
//#include <M5Core2.h>
//#include <M5Unified.h>
void setup() {
  Serial.begin(115200);
  //M5.begin();
  Serial.println("f(setup)");

}
void loop() {
  //M5.update();
}
```
__
It thorws this exception:

ERROR A stack overflow in task IDLE1 has been detected.
abort() was called at PC 0x40088bfc on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40088974:0x3ffba560 0x40088be5:0x3ffba580 0x40088bfc:0x3ffba5a0 0x4008a4de:0x3ffba5c0 0x4008be74:0x3ffba5e0 0x4008be2a:0xa5a5a5a5

Rebooting...
__

Tinyu-Zhao, 2023-04-02:
Obviously your simplest program is wrong, you did not include the header file Arduino.h

You need to add #include <Arduino.h> at the top

Tinyu-Zhao, 2023-04-02:
You can use [M5Burner](https://docs.m5stack.com/en/download) to download the M5Core2 factory test program to check if the M5Core2 is damaged.

fbritop, 2023-04-02:
Thanks
But the problem relies that m5 boards library after an update is showing 2
boards. One of which does not work

You should check the board definition file for Arduino I think

Cheers

Tinyu-Zhao, 2023-04-02:
In addition, M5Core2 and M5Unified cannot be used together, because they are both libraries that control the host device and can cause conflicts

M5Core2 library is specifically for M5Core2.
M5Unified can basically control all the M5 host devices.

fbritop, 2023-04-02:
[image: image.png]

Tinyu-Zhao, 2023-04-02:
> [image: image.png]
> 
> 
> > In addition, M5Core2 and M5Unified cannot be used together, because they
> > are both libraries that control the host device and can cause conflicts
> >
> > M5Core2 library is specifically for M5Core2.
> > M5Unified can basically control all the M5 host devices.
> >
> >
> 
> 
> Felipe Eduardo Brito Pearson
> Ingeniero Informático

I can't see anything.

closed 2024-07-01 by Tinyu-Zhao