m5stack/M5Core2 #127: Boot loop on M5Core2 after burning Core2factoryTest

sixhills, 2023-04-10:
### Describe the bug

Using M5Burner to re-install Core2FactoryTest on a new Core2 device results in the device entering an infinite boot loop.

### To reproduce

1. In M5Burner, slect "Core2 & Tough" from the left-hand pane.
2. in the right-hand pane, scroll down to find Core2FactoryTest and click "Download" then "Burn".
3. The burn appears to work and the device resets.
4. The screen on the Core2 remains blank.
5. Selecting "Serial Monitor" from the top menu shows the device in an infinite boot loop with an illegal instruction panic.

### Expected behavior

Expoected behaviour after a successful burn is for the Core2Factorytest program to start.

### Screenshots

Serial monitor output:

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1100
ho 0 tail 12 room 4
load:0x40078000,len:12308
load:0x40080400,len:3076
entry 0x400805ec
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400f2f20: 443a20c2 0d0e272a 02a8130c
....

### Environment

- OS: Windows 10

### Additional context

Burning UIFlow to the Core2 works, as do some of the other firmware programs listed in M5Burner.  But other firmware programs listed in M5Burner (for example, M5HamRadioCore2) also fail, entering an infinite boot loop after the device resets.  This is repeatable: programs that work always work; programs that fail always fail.

There have now been three reports of this problem with new Core2 devices in the community forum.  Perhaps a recent hardware change has caused the burn process to fail?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

sixhills, 2023-04-15:
Armel has correctly deduced the cause of this issue, which is that only the latest version of the ESP-IDF SDK developed by Espressif is compatible with new Core2 devices.  Firmware built using the earlier version of the ESP-IDK SDK doesn't work on new Core2 devices and enters an infinite loop on booting, reporting either IllegalInstruction or InstrFetchProhibited.

This means that about half of the firmware listed in M5Burner will need to be rebuilt by their authors using the latest SDK to make them work with new Core2s.

closed 2023-04-15 by sixhills

ortlof, 2023-07-31:
On wich update will the Arudio Library Support the new ESP-IDK SDK i have a batch of new Core2s wich have the same bootloop on 1.0.5 when an old one works fine.