m5stack/M5Core2 #128: All LEDs turning on

capedra, 2023-04-21:
I've been using M5Stack Core2 AWS and after it's been running for several hours, all the Neopixel LEDs turns on and they even change colors whenever I press some key on the screen. Also, ```M5.Axp.SetPeripherialsPower(false); // Turn on all external peripherials (5V power)``` doesn't seem to turn them off and I can only turn them off using FastLED library by putting 
```
  FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS); // GRB ordering is assumed
  FastLED.setBrightness(0);
  leds[0] = CRGB::Black;
  leds[1] = CRGB::Black;
  leds[2] = CRGB::Black;
  leds[3] = CRGB::Black;
  leds[4] = CRGB::Black;
  leds[5] = CRGB::Black;
  leds[6] = CRGB::Black;
  leds[7] = CRGB::Black;
  leds[8] = CRGB::Black;
  leds[9] = CRGB::Black;
  FastLED.show();
```
Please help. Thanks!

Tinyu-Zhao, 2024-06-07:
Can you provide an example of what you were using so that I can more easily reproduce the problem?

Tinyu-Zhao, 2024-07-03:
This matter has been in the past for too long and has not been answered by the questioner, so the question is closed.

closed 2024-07-03 by Tinyu-Zhao