m5stack/M5Core2 #134: Arduino IDE not finding MODULE_GRBL13.2

capedra, 2023-09-06:
Whenever I try to install this library on Arduino IDE, I get the following error:

[image]

I guess it should look for ```MODULE_GRBL_13.2``` instead of ```MODULE_GRBL13.2```

kartben, 2023-09-07:
I am having the same issue.

capedra, 2023-09-10:
@Tinyu-Zhao 
@tobozo 
somebody could easily fix it in my humble opinion.

tobozo, 2023-09-11:
@capedra until [the PR](https://github.com/m5stack/M5Core2/pull/135) is merged and a new release is produced, you can choose to install the M5Core2 library without dependencies to skip that error message.

The dependencies for M5Core2 (M5Family,M5Module-4Relay,Module_GRBL_13.2,Ethernet2) can still be installed separately.

tobozo, 2023-09-22:
solved in latest release

closed 2024-06-24 by Tinyu-Zhao