m5stack/M5Core2 #140: error: conflicting declaration 'using EncodeRange = struct lgfx::v1::EncodeRange'  using EncodeRange = lgfx::EncodeRange;

GavinBeattie, 2023-11-02:
### Describe the bug

Error on compile in Arduino IDE that wasn't present before, guessing that Arduino auto library update triggered issue.
Traced as far as:
#include <M5Core2.h> // No error with M5Unified.h

Error msg:
In file included from /Users/.../Documents/Arduino/core2-base-iot/core2-base-iot.ino:4:
/Users/.../Documents/Arduino/libraries/M5GFX/src/M5GFX.h:263:38: error: conflicting declaration 'using EncodeRange = struct lgfx::v1::EncodeRange'
 using EncodeRange = lgfx::EncodeRange;
                                      ^
In file included from /Users/.../Documents/Arduino/libraries/M5Core2/src/utility/In_eSPI.h:480,
                 from /Users/.../Documents/Arduino/libraries/M5Core2/src/M5Display.h:10,
                 from /Users/.../Documents/Arduino/libraries/M5Core2/src/M5Core2.h:12,
                 from /Users/.../Documents/Arduino/core2-base-iot/core2-base-iot.ino:3:
/Users/.../Documents/Arduino/libraries/M5Core2/src/Fonts/GFXFF/gfxfont.h:26:3: note: previous declaration as 'typedef struct EncodeRange EncodeRange'
 } EncodeRange;
   ^~~~~~~~~~~

### To reproduce

Error on compile that wasn't present before, guessing that Arduino auto library update triggered issue.
#include <M5Core2.h> // No error with M5Unified.h

In file included from /Users/.../Documents/Arduino/core2-base-iot/core2-base-iot.ino:4:
/Users/gavinbeattie/Documents/Arduino/libraries/M5GFX/src/M5GFX.h:263:38: error: conflicting declaration 'using EncodeRange = struct lgfx::v1::EncodeRange'
 using EncodeRange = lgfx::EncodeRange;
                                      ^
In file included from /Users/gavinbeattie/Documents/Arduino/libraries/M5Core2/src/utility/In_eSPI.h:480,
                 from /Users/gavinbeattie/Documents/Arduino/libraries/M5Core2/src/M5Display.h:10,
                 from /Users/gavinbeattie/Documents/Arduino/libraries/M5Core2/src/M5Core2.h:12,
                 from /Users/gavinbeattie/Documents/Arduino/core2-base-iot/core2-base-iot.ino:3:
/Users/gavinbeattie/Documents/Arduino/libraries/M5Core2/src/Fonts/GFXFF/gfxfont.h:26:3: note: previous declaration as 'typedef struct EncodeRange EncodeRange'
 } EncodeRange;
   ^~~~~~~~~~~

### Expected behavior

No errors

### Environment

- OS: macOS
- IDE &IDE Version: Arduino IDE
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

andrea-casalicchio-crs, 2023-12-06:
Exactly the same error here:

Environment #1
OS: macOS
IDE &IDE Version: PlaformIO Core 6.1.11

Environment #2
OS: macOS
IDE &IDE Version: Arduino IDE Latest

Environment #3
OS: Windows 11
IDE &IDE Version: PlaformIO Core 6.1.11

Clement-HU, 2024-02-19:
Environment that is possible to compile:
libraries' versions: M5Core2-**v0.1.3** and M5GFX-**v0.1.8**
OS:Windows 10
IDE&IDE Version:Arduino-**v2.0.0**
Man those versions are are all too old!

Clement-HU, 2024-02-19:
So newer versions seem to be buggier?

lbuque, 2024-05-24:
@GavinBeattie I installed M5Core2 0.1.9 and no errors were encountered during compilation. Please provide test code.

closed 2024-06-24 by Tinyu-Zhao