m5stack/M5Core2 #142: IMU (MPU6886) Library Expansion

JasonPittenger, 2023-11-17:
### Describe the bug

Can you expand the IMU library to incorporate more features of the MPU6886?

It has a built in low pass filter on both the gyro and accel data.  
This is useful to reduce the needed sampling rate without getting aliasing errors.
For example, I set it to the Accel and Gyro low pass filter to 5.1 Hz and 5.0 Hz respectively.
This reduces noise and means I can sample at a much lower rate without high frequency data messing with the results.
I am doing this manually right now.

The accelerometer offers low pass frequencies of 5.1, 10.2, 21.2, 44.8, 99.0, 218.1, 420.0 and 1046 Hz.
(page 37-38 of https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/MPU-6886-000193%2Bv1.1_GHIC_en.pdf)
The Gyroscope offers low pass frequencies of 5, 10, 20, 41, 92, 176, 250, 3281, 8173 Hz.
(page 36 of https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/MPU-6886-000193%2Bv1.1_GHIC_en.pdf)

There is also a note in the data sheet that a bit in one of the registers should be set to 1 on every power up.
(page 45 of https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/MPU-6886-000193%2Bv1.1_GHIC_en.pdf)
This should be included in the software library initialization sequence.

```
		//Set IMU accelerometer LPF frequency to 5.1Hz.
		Wire1.beginTransmission(MPU6886_ADDRESS);
		Wire1.write(MPU6886_ACCEL_CONFIG2);
		Wire1.write(0x06);
		Wire1.endTransmission();
		delay(1);
		//Set IMU Gyro LPF frequency to 5Hz.
		Wire1.beginTransmission(MPU6886_ADDRESS);
		Wire1.write(MPU6886_CONFIG);
		Wire1.write(0x06);
		Wire1.endTransmission();
		delay(1);
		//To avoid limiting sensor output to less than 0x7F7F, set this bit to 1. This should be done every time the MPU-6886 is powered up.
		Wire1.beginTransmission(MPU6886_ADDRESS);
		Wire1.write(MPU6886_ACCEL_INTEL_CTRL);
		Wire1.write(0x02);
		Wire1.endTransmission();
		delay(1);
```

Lastly, the MPU6886 has a low power mode that would be nice to have included in the library.
(page 47 of https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/MPU-6886-000193%2Bv1.1_GHIC_en.pdf)

```
//Set IMU sleep to 1 and disable temp sensor.
Wire1.beginTransmission(MPU6886_ADDRESS);
Wire1.write(MPU6886_PWR_MGMT_1);
Wire1.write(0x49);
Wire1.endTransmission();

//Wake up the IMU
Wire1.beginTransmission(MPU6886_ADDRESS);
Wire1.write(MPU6886_PWR_MGMT_1);
Wire1.write(0x01);
Wire1.endTransmission();
```

### To reproduce

Any usage of the MPU6886

### Expected behavior

I expect the library to allow setting the low pass filters, not limit the output to 0x7F7F, and allow low power/sleep mode.

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

icyqwq, 2024-05-22:
Feature added at commit 4492379.

closed 2024-05-22 by icyqwq

apwebber, 2025-07-17:
Seconded, I'm looking for this for the M5Stack