m5stack/M5Core2 #146: I2S error with M5Core2 V0.1.5 but not with V0.1.3

NDC-SC, 2024-01-11:
### Describe the bug

With M5Core2 V0.1.5 library, accessing I2S produces the error messages:

E (1208) I2S: i2s_driver_uninstall(2048): I2S port 0 has not installed
E (1245) I2S: register I2S object to platform failed
E (1245) I2S: i2s_read(2265): RX mode is not enabled

but no error with M5Core2 V0.1.3 library and the sketch runs properly

### To reproduce

// sketch provided below

#include <M5Core2.h>          // include M5Core2 and
#include <driver/i2s.h>         //    ESP32 I2S libraries
const int bufferLen = 1024;       // DMA buffer size (samples)
int N = 1, yData[320], yBase = 120;
int minY = 0, maxY = 2000;        // min and max values M5GO
//int minY = -600, maxY = 600;       // min and max values M5Stack

void setup()
{
  M5.begin();
  M5.Lcd.fillScreen(BLACK);       // initialise display
  M5.Lcd.setTextSize(1);
  M5.Lcd.setTextColor(WHITE, BLACK);
  M5.Lcd.drawString("PDM mic", 50, 0, 4);
  for (int i=0; i<320; i++) yData[i] = yBase;
  I2Sconfig();            // configure I2S
  pinConfig();            //    and GPIO pins
}

void I2Sconfig()            // function to configure I2S
{
  i2s_config_t i2s_config = {       // receive and PDM modes
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM),
    .sample_rate = 44100,         // sample frequency
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // 16-bit sampling
    .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT, // mono channel sampling
    .communication_format = I2S_COMM_FORMAT_STAND_I2S,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 2,           // DMA buffers
    .dma_buf_len = bufferLen          // DMA buffer length
  };
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
}

void pinConfig()            // function to configure I2S pins
{
  i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_PIN_NO_CHANGE,    // bit clock frequency
    .ws_io_num = 0,         // word select (left /right) clock
    .data_out_num = I2S_PIN_NO_CHANGE,    // data output
    .data_in_num = 34         // data input
  };
  i2s_set_pin(I2S_NUM_0, &pin_config);
}

void loop()
{
  wave();             // call wave function
}

void wave()       // function to read DMA buffer and display waveform
{
  size_t bits = 0;
  int16_t buffer[bufferLen] = {0};      // define and read I2S data buffer
  i2s_read(I2S_NUM_0, &buffer, sizeof(buffer), &bits, portMAX_DELAY);
  int bytes = bits / 8;         // convert bits to bytes
  if(bytes > 0)
  {
    for (int i=0; i<bytes; i=i+N)
    {
      M5.Lcd.drawLine(0, yData[0], 1, yData[1], BLACK); // overwrite value
      for (int j=1; j<320; j++) yData[j-1] = yData[j];     // shift one position
      int temp = constrain(buffer[i], minY, maxY);   // constrain buffer values
      yData[319] = map(temp, minY, maxY, 30, 240);      // map to LCD height
      for (int j=1; j<319; j++)
      {
        M5.Lcd.drawLine(j, yData[j-1], j+1, yData[j], BLACK); // overwrite and 
        M5.Lcd.drawLine(j, yData[j],   j+1, yData[j+1], RED); // draw new line
      }
    }
  }
  M5.update();
}

### Expected behavior

M5Core2 screen displays waveform of sound received by built-in microphone

### Screenshots

[image]

### Environment

- OS:  Windows 10
- IDE &IDE Version:    Arduino 1.8.19
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

NDC-SC, 2024-01-12:
The issue arose from an error message displayed with the M5Core2 V0.1.8 library.
The V1.0.7, V1.0.5 and V1.0.3 libraries were separately loaded and the problem was "resolved" by the V1.0.3 library.

The latest ESP32 Board Manager V2.0.14 is installed

GOB52, 2024-05-23:
This is due to the addition of Speaker introduced in V0.1.4, which causes I2S-related processing in M5.begin.

Calling begin in this way can suppress I2S-related processing.
(Set the sixth argument to false)

```cpp
M5.begin(
true, // Enable LCD
true, // Enable SD
true, // Enable Serial
true, // Enable I2C
mbus_mode_t::kMBusModeOutput, // MBus mode
false // Disable Speaker  <<<< HERE!
);
```

GOB52, 2024-05-23:
Results in V0.1.9

[image]

NDC-SC, 2024-05-23:
Thanks for the clarification.