m5stack/M5Core2 #149: Play wav file from SD card

Muhlik2018, 2024-04-16:
Hi,

I'm trying to play wav file from SD card on my Core 2, I find the M5Unified library seems to have the solution however M5Core2 Library does not. 
Wondering if you can update the speaker class in M5 Core2 library to the same as M5Unified library, or is any solution can be provided under current M5Core2 library?

Thanks

GOB52, 2024-05-23:
It is recommended to use M5Unified.
See also https://github.com/m5stack/M5Unified/issues/29

The M5Core2 Speaker can also play in this way, although the format is limited.

https://github.com/m5stack/M5Core2/blob/develop/examples/Basics/speak_sd/speak_sd.ino

GOB52, 2024-05-30:
Hi @Muhlik2018 
What do you think about this?

> It is recommended to use M5Unified. See also [m5stack/M5Unified#29](https://github.com/m5stack/M5Unified/issues/29)
> 
> The M5Core2 Speaker can also play in this way, although the format is limited.
> 
> https://github.com/m5stack/M5Core2/blob/develop/examples/Basics/speak_sd/speak_sd.ino

Tinyu-Zhao, 2024-07-02:
The issue has been resolved in PR #151

closed 2024-07-02 by Tinyu-Zhao