m5stack/M5Core2 #154: M5Core2 build error AXP.h:34:37: error: 'uint8_t' in namespace 'std' does not name a type; did you mean 'wint_t'?

lboue, 2024-11-08:
### Describe the bug

Build [Basics/display/display.ino](https://github.com/m5stack/M5Core2/blob/master/examples/Basics/display/display.ino)

**M5Core2 build error**
```
In file included from c:\Users\user1\Documents\Arduino\libraries\M5Core2\src/M5Core2.h:19:
c:\Users\user1\Documents\Arduino\libraries\M5Core2\src/AXP.h: At global scope:
c:\Users\user1\Documents\Arduino\libraries\M5Core2\src/AXP.h:34:37: error: 'uint8_t' in namespace 'std' does not name a type; did you mean 'wint_t'?
   34 | bool writeRegister8Array(const std::uint8_t *reg_data_array,
      |                                     ^~~~~~~
      |                                     wint_t
```

### To reproduce

Build [Basics/display/display.ino](https://github.com/m5stack/M5Core2/blob/master/examples/Basics/display/display.ino)

### Expected behavior

Build example

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5Core2/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-11-11:
The ESP32 version needs to be below 3.x, you can use this https://github.com/espressif/arduino-esp32/releases/tag/2.0.17

markab, 2024-11-15:
Hello, I have the same problem, I have downgraded to 2.0.17 and it compiles but with errors, when will you support esp version 3?

Tinyu-Zhao, 2024-11-28:
After waiting for M5Unified to fully adapt to IDF 5.x.x, we will support esp verison 3.

> Hello, I have the same problem, I have downgraded to 2.0.17 and it compiles but with warnings, when will you support esp version 3?

After waiting for M5Unified to fully adapt to IDF 5.x.x, we will support esp verison 3.After waiting for M5Unified to fully adapt to IDF 5.x.x, we will support esp verison 3.

closed 2024-11-28 by Tinyu-Zhao