m5stack/M5Core2 #158: display changes 0.1.5 -> 0.2.0

DrJaymz, 2024-12-17:
I have been using a CORE2 with green LED for a couple of years and my code functions fine.  I note that the library version is 0.1.5.
I just got a new CORE2 with a blue LED and wrote my built code as-is to it and it is working fine according to the debug output, touchscreen etc.  but the screen is black, no backlight and can't tell if the screen is actually displaying anything.

Perhaps this is to do with the change to the AXP192?  I'm reluctant to upgrade by old one to 0.2.0 are there any other breaking changes?

Tinyu-Zhao, 2024-12-17:
All changes are [recorded](https://github.com/m5stack/M5Core2/compare/0.1.5...master) on github. The time span is too long and I can’t answer your last question.

Your new CORE2 should be the AXP2101 version, so it cannot light up the screen and can only use the latest library.

DrJaymz, 2024-12-17:
> All changes are [recorded](https://github.com/m5stack/M5Core2/compare/0.1.5...master) on github. The time span is too long and I can’t answer your last question.
> 
> Your new CORE2 should be the AXP2101 version, so it cannot light up the screen and can only use the latest library.

Thanks, I have updated to 0.2.0 and the screen is working so thats awesome.  

Then it crashes, but that will be some other problem...  And appears to be something I am doing with audio.  I don't remember audo changes on the changelog.

Backtrace: 0x400e4487:0x3ffb21b0 0x400ddf72:0x3ffb21f0 0x400d3e09:0x3ffb2240 0x400e24e6:0x3ffb2290
  #0  0x400e4487 in i2s_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:2129 (discriminator 2)
  #1  0x400ddf72 in soundsBeep at lib/foo/Core2_Sounds.c:473
  #2  0x400d3e09 in setup() at src/main.cpp:169 (discriminator 4)
  #3  0x400e24e6 in loopTask(void*) at C:/platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
  
  Which is this line:
  
    // Write to internal I2S speaker driver, then free memory allocated
  esp_err_t err = i2s_write(SPEAKER_I2S_NUMBER, raw, length, &bytes_written, portMAX_DELAY);

DrJaymz, 2024-12-17:
Fixed it, now requires M5.Spk.Begin() which I didn't need before and also   M5.Axp.SetSpkEnable(true); again, didn't need that before.

Now it beeps and is horribly distorted and terribly clicky - which means its working as well as the older one did - so all fine!

Tinyu-Zhao, 2024-12-18:
> Fixed it, now requires M5.Spk.Begin() which I didn't need before and also M5.Axp.SetSpkEnable(true); again, didn't need that before.
> 
> Now it beeps and is horribly distorted and terribly clicky - which means its working as well as the older one did - so all fine!

Glad to see you solved the problem 😊.

closed 2024-12-18 by Tinyu-Zhao