m5stack/M5Core2 #159: Battery Example?

DrJaymz, 2024-12-18:
Using the V1.1 with 0.2.0 the battery doesn't appear to work at all.  It did not charge on ANY usb power source.
It will not turn on without USB power.

In my loop I have:
`
  if (M5.Axp.isCharging())
  {
    M5.lcd.clearDisplay(TFT_BLACK);

    M5.Lcd.setCursor(0, 0);
    M5.Lcd.println("Charging...");
    M5.Lcd.printf("%0.2f% (%0.0fma) %0.2fV \n", M5.Axp.GetBatteryLevel(), M5.Axp.GetBatChargeCurrent(), M5.Axp.GetBatVoltage());
    SleepProcessor(500000); // low power sleep for 0.5 sec
    return;
  }
`
On 0.1.5 this showed something like 68.00% 96.6mA 4.16V
On 0.2.0 it shows 67.93 (-1mA) 3.80V 

I didn't see a battery example and to be honest, charging the older one has always been a bit hit and miss, I thought I charged it only to find it flame out after a random time between 5 mins and 6 hours.

I suppose a suitable example would be like charging a phone, you plug it in, if on USB it shoudl show the charging page, otherwise show a normal page and the battery charge level.

Again, I may be victim to changed between the versions, but if there is documentation I can't easily find it.

DrJaymz, 2024-12-18:
So charge current doesn't work.  And charging is always true:

```
bool AXP::isCharging() {
    if (_pmic == pmic_axp192) {
        return axp192.isCharging();
    }
    return true;
}
```

If its an axp192 then work properly otherwise just lie?
Come on guys, what are you playing at?  How to tell if its on charge?

DrJaymz, 2024-12-18:
I see battery current is now in amps rather than milliamps.  An example of how to use the settings, like the charge current etc. would be very useful.

felmue, 2024-12-30:
Hello @DrJaymz 

you could try to use `M5Unified` library instead which supports many (if not all) M5Stack devices.

I've created a battery monitor example for M5Core2 v1.1 [here](https://github.com/felmue/MyM5StackExamples/blob/main/M5Core2/BatteryMonitor/BatteryMonitor.ino).

Thanks
Felix

DrJaymz, 2024-12-31:
> Hello @DrJaymz
> 
> you could try to use `M5Unified` library instead which supports many (if not all) M5Stack devices.
> 
> I've created a battery monitor example for M5Core2 v1.1 [here](https://github.com/felmue/MyM5StackExamples/blob/main/M5Core2/BatteryMonitor/BatteryMonitor.ino).
> 
> Thanks Felix

Thanks I will try that, what is the guidance on using the unified library?  

I can't find anything that says if you're using M5Core2 library you can move to trhe unified library but the following things may require extra work.....

Going back to my initial issue, I just looked at the m5core2 library and its not really fit for purpose, its like they just tried to get rid of the compiler errors and called it fixed.  

To get charging status: M5.Axp.isACIN();
For current I used: M5.Axp.GetBatCurrent() * 1000;

Note that current is in amps now not milliamps as before, which is why my old code which was a sprintf("%0.1f",M5.Axp.GetBatCurrent()) showed nothing at all.

DrJaymz, 2025-01-02:
> Hello @DrJaymz
> 
> you could try to use `M5Unified` library instead which supports many (if not all) M5Stack devices.
> 
> I've created a battery monitor example for M5Core2 v1.1 [here](https://github.com/felmue/MyM5StackExamples/blob/main/M5Core2/BatteryMonitor/BatteryMonitor.ino).
> 
> Thanks Felix

I tried this example and looks to work fine.  I then had a play with the charge current and finally have a reasonable charge current so all in all that works fine.  Then for my major project I has a look at what might have broken, RTC clock (a few changes required but worked), audio not working but I didn't try very hard yet.  I also need to check that the power saving stuff I implemented works but all in all I would probably use the unified library as it feels more complete.

One issue I have been having is poor tft_espi support which is a bit of a dodgy implementation - so that will be my focus.  As its a complex library with great support from bodmer it really needs to be included in a maintainable fashion.