m5stack/M5Core2 #160: Not able do compile anymore

minou65, 2024-12-19:
Hi all
After I have updated arduinoIDE and all librarys I am not able to compile anymore my program. Is thre a know issue or do I have overseen a requirement?

speaker.cpp: In member function bool Speaker::InitI2SSpeakOrMic(int)
 
speaker.cpp: 31:31: error: 'I2S_MCLK_MULTIPLE_DEFAULT' was not declared in this scope; did you mean 'I2S_MCLK_MULTIPLE_512'?
   31 |         .mclk_multiple      = I2S_MCLK_MULTIPLE_DEFAULT
   |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
   |                               I2S_MCLK_MULTIPLE_512
 
M5Display.h:10: In file included from
m5core2.h:12: from
m5touch.cpp:1: from
M5Display.h: In member function void M5Display::startWrite()
 
In_eSPI.h: 233:5: error: 'GPIO' was not declared in this scope
   233 |     GPIO.out_w1tc = (1 << TFT_CS)
   |     ^~~~
M5Display.h:54: note  in expansion of macro CS_L
   54 |         CS_L
   |         ^~~~
M5Display.h: In member function void M5Display::endWrite()
 
In_eSPI.h: 235:14: error: 'GPIO' was not declared in this scope
   235 | #define CS_H GPIO.out_w1ts = (1 << TFT_CS)  \\;GPIO.out_w1ts = (1 << TFT_CS)
   |              ^~~~
 
M5Display.h:66: note  in expansion of macro CS_H
M5Display.h:10: In file included from
   66 |         CS_H
M5Core2.h:12: from
   |         ^~~~
m5core2.cpp:5: from
M5Display.h: In member function void M5Display::startWrite()

minou65, 2024-12-19:
The part with In_eSPI and GPIO not declared is fixed with this MR
https://github.com/m5stack/M5Core2/pull/157

Tinyu-Zhao, 2024-12-20:
Isn’t the content fixed by this PR already included in the latest program? 
[image]

minou65, 2024-12-20:
Hi @Tinyu-Zhao 
Yes, it is. I deleted the library and reinstalled it, which fixed the issue. However, the other errors still exist.

```
speaker.cpp: In member function bool Speaker::InitI2SSpeakOrMic(int)
 
speaker.cpp: 31:31: error: 'I2S_MCLK_MULTIPLE_DEFAULT' was not declared in this scope; did you mean 'I2S_MCLK_MULTIPLE_512'?
   31 |         .mclk_multiple      = I2S_MCLK_MULTIPLE_DEFAULT
   |                               ^~~~~~~~~~~~~~~~~~~~~~~~~
   |                               I2S_MCLK_MULTIPLE_512
 
m5display.cpp: In function bool jpgDecode(jpg_file_decoder_t*, unsigned int (*)(JDEC*, uint8_t*, unsigned int))
 
m5display.cpp: 378:32: error: invalid conversion from 'uint32_t (*)(JDEC*, void*, JRECT*)' {aka 'long unsigned int (*)(JDEC*, void*, JRECT*)'} to 'UINT (*)(JDEC*, void*, JRECT*)' {aka 'unsigned int (*)(JDEC*, void*, JRECT*)'} [-fpermissive]
   378 |     jres = jd_decomp(&decoder, jpgWrite, (uint8_t)jpeg->scale)
   |                                ^~~~~~~~
   |                                |
   |                                uint32_t (*)(JDEC*, void*, JRECT*) {aka long unsigned int (*)(JDEC*, void*, JRECT*)}
 
m5display.cpp:223: In file included from
tjpgd.h:92: note    initializing argument 2 of JRESULT jd_decomp(JDEC*, UINT (*)(JDEC*, void*, JRECT*), BYTE)
   92 | JRESULT jd_decomp (JDEC*, UINT(*)(JDEC*,void*,JRECT*), BYTE)
   |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
m5display.cpp: In member function void M5Display::drawJpg(const uint8_t*, size_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, jpeg_div_t)
 
m5display.cpp: 416:22: error: invalid conversion from 'uint32_t (*)(JDEC*, uint8_t*, uint32_t)' {aka 'long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)'} to 'unsigned int (*)(JDEC*, uint8_t*, unsigned int)' {aka 'unsigned int (*)(JDEC*, unsigned char*, unsigned int)'} [-fpermissive]
   416 |     jpgDecode(&jpeg, jpgRead)
   |                      ^~~~~~~
   |                      |
   |                      uint32_t (*)(JDEC*, uint8_t*, uint32_t) {aka long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)}
m5display.cpp:352: note    initializing argument 2 of bool jpgDecode(jpg_file_decoder_t*, unsigned int (*)(JDEC*, uint8_t*, unsigned int))
   352 |                       unsigned int (*reader)(JDEC *, uint8_t *, unsigned int)) {
   |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
m5display.cpp: In member function void M5Display::drawJpgFile(fs::FS&, const char*, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, jpeg_div_t)
 
m5display.cpp: 454:22: error: invalid conversion from 'uint32_t (*)(JDEC*, uint8_t*, uint32_t)' {aka 'long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)'} to 'unsigned int (*)(JDEC*, uint8_t*, unsigned int)' {aka 'unsigned int (*)(JDEC*, unsigned char*, unsigned int)'} [-fpermissive]
   454 |     jpgDecode(&jpeg, jpgReadFile)
   |                      ^~~~~~~~~~~
   |                      |
   |                      uint32_t (*)(JDEC*, uint8_t*, uint32_t) {aka long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)}
m5display.cpp:352: note    initializing argument 2 of bool jpgDecode(jpg_file_decoder_t*, unsigned int (*)(JDEC*, uint8_t*, unsigned int))
   352 |                       unsigned int (*reader)(JDEC *, uint8_t *, unsigned int)) {
   |                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
m5display.cpp: In member function void M5Display::drawPngUrl(const char*, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, uint16_t, double, uint8_t)
 
m5display.cpp: 571:9: error: 'WiFi' was not declared in this scope
   571 |     if (WiFi.status() != WL_CONNECTED) {
   |         ^~~~
 
m5display.cpp: 571:26: error: 'WL_CONNECTED' was not declared in this scope
   571 |     if (WiFi.status() != WL_CONNECTED) {
   |                          ^~~~~~~~~~~~
 
m5display.cpp: 585:5: error: 'WiFiClient' was not declared in this scope
   585 |     WiFiClient *stream = http.getStreamPtr()
   |     ^~~~~~~~~~
 
m5display.cpp: 585:17: error: 'stream' was not declared in this scope; did you mean 'std::io_errc::stream'?
   585 |     WiFiClient *stream = http.getStreamPtr()
   |                 ^~~~~~
   io_errc::stream
 
ios:42: In file included from
ostream:38: from
unique_ptr.h:41: from
memory:76: from
fs.h:24: from
M5Display.h:6: from
m5display.cpp:1: from
ios_base.h:204: note  std  io_errc  stream declared here
   204 |   enum class io_errc { stream = 1 }
   |                        ^~~~~~
 
sprite.cpp: In function bool jpgDecode(jpg_file_decoder_t*, uint32_t (*)(JDEC*, uint8_t*, uint32_t))
 
sprite.cpp: 2342:41: error: invalid conversion from 'uint32_t (*)(JDEC*, uint8_t*, uint32_t)' {aka 'long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)'} to 'UINT (*)(JDEC*, BYTE*, UINT)' {aka 'unsigned int (*)(JDEC*, unsigned char*, unsigned int)'} [-fpermissive]
   2342 |     JRESULT jres = jd_prepare(&decoder, reader, work, 3100, jpeg)
   |                                         ^~~~~~
   |                                         |
   |                                         uint32_t (*)(JDEC*, uint8_t*, uint32_t) {aka long unsigned int (*)(JDEC*, unsigned char*, long unsigned int)}
 
sprite.cpp:2208: In file included from
tjpgd.h:91: note    initializing argument 2 of JRESULT jd_prepare(JDEC*, UINT (*)(JDEC*, BYTE*, UINT), void*, UINT, void*)
[... cut ...] void*, JRECT*)'} [-fpermissive]
   2364 |     jres = jd_decomp(&decoder, jpgWrite, (uint8_t)jpeg->scale)
   |                                ^~~~~~~~
   |                                |
   |                                uint32_t (*)(JDEC*, void*, JRECT*) {aka long unsigned int (*)(JDEC*, void*, JRECT*)}
tjpgd.h:92: note    initializing argument 2 of JRESULT jd_decomp(JDEC*, UINT (*)(JDEC*, void*, JRECT*), BYTE)
   92 | JRESULT jd_decomp (JDEC*, UINT(*)(JDEC*,void*,JRECT*), BYTE)
   |    
```

Tinyu-Zhao, 2024-12-23:
Which board management are you using?   
[image]

minou65, 2024-12-23:
Hi @Tinyu-Zhao 
Thanks for comming back. I use esp32

Tinyu-Zhao, 2024-12-24:
> Hi @Tinyu-Zhao Thanks for comming back. I use esp32

There will be no problem using M5Stack. Specific tutorials can be found here.
https://docs.m5stack.com/en/arduino/arduino_board